Lora-net/sx1302_hal #10: After stop, reconfig and start: Failed to set SX1250_0 in STANDBY_RC mode

brocaar, 2020-04-17:
I'm working on an application which implements the SX1301/2 HAL as a daemon. One of the features is to re-configure the concentrator channel-plan, without restarting the daemon. The initial startup works fine, however stopping the SX1302 (`lgw_stop`) doing the new configuration and starting (`lgw_start`') fails. In short I execute the following sequence of commands:

1. board_setconf
2. timestamp_setconf
3. txgain_setconf
4. rxrf_setconf
5. rxif_setconf
6. start

When there is new configuration to apply, I execute `stop` and start again with 1) (with the new configuration). However, step 6) fails with:

```
ERROR: Failed to set SX1250_0 in STANDBY_RC mode
ERROR: Failed to set SX1250_1 in STANDBY_RC mode
ERROR: Failed to load fw
```

I'm unable to restart my binary after this failure, unless I execute the [reset script](https://github.com/Lora-net/sx1302_hal/blob/master/tools/reset_lgw.sh) first.

The same sequence with the SX1301 HAL works fine and there is no need to reset the SX1301 reset script.

For the SX1302 I'm using the v1.0.5 HAL version.

wong-hao, 2020-10-08:
[image]
[image]
I have the same question. Only two consecutive runs of the same script can guarantee the second successful operation

ingmanuelalfonso, 2020-11-19:
Hi, do you solve that?. I am with the same error in my gateway. The SPI, i2c tests passed but the hal_rx and tx give me the same error you report.

mightChamp, 2021-03-30:
ERROR: Failed to set SX1250_0 in STANDBY_RC mode
ERROR: Failed to set SX1250_1 in STANDBY_RC mode

if I bypass this error in code, then everything works fine.
Is any significance of this these errors?

nilesh-dryad, 2021-05-06:
If you are not using SX1261, then make SX1261_NRESET and SX1261_NSS pin high(3.3V) permanently and check again.

Gazella019, 2021-06-21:
In file sx1250_spi.c/ sx1250_spi_r funtion
Before copying the data in SPI buffer to input data array, i add a delay, then it works
`wait_ms(500);
memcpy(data, in_buf + cmd_size, size);`

wong-hao, 2021-11-28:
> ERROR: Failed to set SX1250_0 in STANDBY_RC mode ERROR: Failed to set SX1250_1 in STANDBY_RC mode
> 
> if I bypass this error in code, then everything works fine. Is any significance of this these errors?

I find a better solution: commit [this code part](https://github.com/Lora-net/sx1302_hal/blob/4b42025d1751e04632c0b04160e0d29dbbb222a5/packet_forwarder/src/lora_pkt_fwd.c#L1942) So that it does not reset when exiting the program.

Benixpearlin, 2022-02-18:
Helo,
i am using raspberry pi with sx1302
i am using sx1302_hal-master packet forwarder code

but i am facing the following error while starting the packet forwarder

*** Packet Forwarder ***
Version: 2.1.0
*** SX1302 HAL library version info ***
Version: 2.1.0;

INFO: Little endian host
INFO: found configuration file global_conf.json.sx1250.CN490, parsing it
INFO: global_conf.json.sx1250.CN490 does contain a JSON object named SX130x_conf, parsing SX1302 parameters
INFO: com_type SPI, com_path /dev/spidev0.0, lorawan_public 1, clksrc 0, full_duplex 0
INFO: antenna_gain 0 dBi
INFO: Configuring legacy timestamp
INFO: no configuration for SX1261
INFO: Configuring Tx Gain LUT for rf_chain 0 with 16 indexes for sx1250
INFO: radio 0 enabled (type SX1250), center frequency 471400000, RSSI offset -207.000000, tx enabled 1, single input mode 1
INFO: radio 1 enabled (type SX1250), center frequency 475000000, RSSI offset -207.000000, tx enabled 0, single input mode 1
INFO: Lora multi-SF channel 0> radio 0, IF -300000 Hz, 125 kHz bw, SF 5 to 12
INFO: Lora multi-SF channel 1> radio 0, IF -100000 Hz, 125 kHz bw, SF 5 to 12
INFO: Lora multi-SF channel 2> radio 0, IF 100000 Hz, 125 kHz bw, SF 5 to 12
INFO: Lora multi-SF channel 3> radio 0, IF 300000 Hz, 125 kHz bw, SF 5 to 12
INFO: Lora multi-SF channel 4> radio 1, IF -300000 Hz, 125 kHz bw, SF 5 to 12
INFO: Lora multi-SF channel 5> radio 1, IF -100000 Hz, 125 kHz bw, SF 5 to 12
INFO: Lora multi-SF channel 6> radio 1, IF 100000 Hz, 125 kHz bw, SF 5 to 12
INFO: Lora multi-SF channel 7> radio 1, IF 300000 Hz, 125 kHz bw, SF 5 to 12
INFO: Lora std channel> radio 1, IF -200000 Hz, 250000 Hz bw, SF 7, Explicit header
INFO: FSK channel> radio 1, IF 300000 Hz, 125000 Hz bw, 50000 bps datarate
INFO: global_conf.json.sx1250.CN490 does contain a JSON object named gateway_conf, parsing gateway parameters
INFO: gateway MAC address is configured to AA555A0000000000
INFO: server hostname or IP address is configured to “localhost”
INFO: upstream port is configured to “1730”
INFO: downstream port is configured to “1730”
INFO: downstream keep-alive interval is configured to 2 seconds
INFO: statistics display interval is configured to 30 seconds
INFO: upstream PUSH_DATA time-out is configured to 100 ms
INFO: packets received with a valid CRC will be forwarded
INFO: packets received with a CRC error will NOT be forwarded
INFO: packets received with no CRC will NOT be forwarded
INFO: GPS serial port path is configured to “/dev/ttyS0”
INFO: Reference latitude is configured to 0.000000 deg
INFO: Reference longitude is configured to 0.000000 deg
INFO: Reference altitude is configured to 0 meters
INFO: Beaconing period is configured to 0 seconds
INFO: Beaconing signal will be emitted at 869525000 Hz
INFO: Beaconing datarate is set to SF9
INFO: Beaconing modulation bandwidth is set to 125000Hz
INFO: Beaconing TX power is set to 14dBm
INFO: Beaconing information descriptor is set to 0
INFO: global_conf.json.sx1250.CN490 does contain a JSON object named debug_conf, parsing debug parameters
INFO: got 2 debug reference payload
INFO: reference payload ID 0 is 0xCAFE1234
INFO: reference payload ID 1 is 0xCAFE2345
INFO: setting debug log file name to loragw_hal.log
INFO: [main] TTY port /dev/ttyS0 open for GPS synchronization
CoreCell reset through GPIO23…
SX1261 reset through GPIO23…
CoreCell power enable through GPIO18…
CoreCell ADC reset through GPIO13…
Opening SPI communication interface
Note: chip version is 0x00 (v0.0)
ERROR: Failed to set SX1250_0 in STANDBY_XOSC mode
ERROR: failed to setup radio 0
ERROR: [main] failed to start the concentrator

what am i missing??

ManuIoT, 2022-06-22:
For me this error was caused for a wrong SX1302 solder. I just soldered again and corrected soldering pins with the help o a microscope.

sankeycoder7, 2024-06-18:
debian@beaglebone:~/lora/sx1302_hal/packet_forwarder$ ./lora_pkt_fwd -c global_conf.json.sx1250.EU868
*** Packet Forwarder ***
Version: 2.1.0
*** SX1302 HAL library version info ***
Version: 2.1.0;
***
INFO: Little endian host
INFO: found configuration file global_conf.json.sx1250.EU868, parsing it
INFO: global_conf.json.sx1250.EU868 does contain a JSON object named SX130x_conf, parsing SX1302 parameters
INFO: com_type SPI, com_path /dev/spidev0.0, lorawan_public 1, clksrc 0, full_duplex 0
INFO: antenna_gain 0 dBi
INFO: Configuring legacy timestamp
INFO: Configuring Tx Gain LUT for rf_chain 0 with 16 indexes for sx1250
INFO: radio 0 enabled (type SX1250), center frequency 867500000, RSSI offset -215.399994, tx enabled 1, single input mode 0
INFO: radio 1 enabled (type SX1250), center frequency 868500000, RSSI offset -215.399994, tx enabled 0, single input mode 0
INFO: Lora multi-SF channel 0>  radio 1, IF -400000 Hz, 125 kHz bw, SF 5 to 12
INFO: Lora multi-SF channel 1>  radio 1, IF -200000 Hz, 125 kHz bw, SF 5 to 12
INFO: Lora multi-SF channel 2>  radio 1, IF 0 Hz, 125 kHz bw, SF 5 to 12
INFO: Lora multi-SF channel 3>  radio 0, IF -400000 Hz, 125 kHz bw, SF 5 to 12
INFO: Lora multi-SF channel 4>  radio 0, IF -200000 Hz, 125 kHz bw, SF 5 to 12
INFO: Lora multi-SF channel 5>  radio 0, IF 0 Hz, 125 kHz bw, SF 5 to 12
INFO: Lora multi-SF channel 6>  radio 0, IF 200000 Hz, 125 kHz bw, SF 5 to 12
INFO: Lora multi-SF channel 7>  radio 0, IF 400000 Hz, 125 kHz bw, SF 5 to 12
INFO: Lora std channel> radio 1, IF -200000 Hz, 250000 Hz bw, SF 7, Explicit header
INFO: FSK channel> radio 1, IF 300000 Hz, 125000 Hz bw, 50000 bps datarate
INFO: global_conf.json.sx1250.EU868 does contain a JSON object named gateway_conf, parsing gateway parameters
INFO: gateway MAC address is configured to AA555A0000000000
INFO: server hostname or IP address is configured to "localhost"
INFO: upstream port is configured to "1730"
INFO: downstream port is configured to "1730"
INFO: downstream keep-alive interval is configured to 10 seconds
INFO: statistics display interval is configured to 30 seconds
INFO: upstream PUSH_DATA time-out is configured to 100 ms
INFO: packets received with a valid CRC will be forwarded
INFO: packets received with a CRC error will NOT be forwarded
INFO: packets received with no CRC will NOT be forwarded
INFO: GPS serial port path is configured to "/dev/ttyS0"
INFO: Reference latitude is configured to 0.000000 deg
INFO: Reference longitude is configured to 0.000000 deg
INFO: Reference altitude is configured to 0 meters
INFO: Beaconing period is configured to 0 seconds
INFO: Beaconing signal will be emitted at 869525000 Hz
INFO: Beaconing datarate is set to SF9
INFO: Beaconing modulation bandwidth is set to 125000Hz
INFO: Beaconing TX power is set to 14dBm
INFO: Beaconing information descriptor is set to 0
INFO: global_conf.json.sx1250.EU868 does contain a JSON object named debug_conf, parsing debug parameters
INFO: got 2 debug reference payload
INFO: reference payload ID 0 is 0xCAFE1234
INFO: reference payload ID 1 is 0xCAFE2345
INFO: setting debug log file name to loragw_hal.log
WARNING: [main] impossible to open /dev/ttyS0 for GPS sync (check permissions)
CoreCell reset through GPIO11...
SX1261 reset through GPIO11...
CoreCell power enable through GPIO18...
CoreCell ADC reset through GPIO13...
Opening SPI communication interface
Note: chip version is 0xFF (v15.15)
freq_hz=867500000
0
ERROR: sx1250 Image Calibration Error
ERROR: radio calibration failed
ERROR: radio calibration failed
ERROR: [main] failed to start the concentrator

I'm facing this is problem give the solution

mcoracin, 2024-06-23:
Hi @brocaar , I know that it is a very old issue, but do you still have an issue ? Did you workaround it somehow ?
I'll look into this issue of start/stop/start.