Lora-net/sx1302_hal #11: is wrong format of pa_gain in global_conf.json.sx1250.CN490?

BillowMai, 2020-04-23:
the pa_gain is from 0 to 3 in global_conf.json.sx1250.CN490, but the pa_gain is limited to 0 or 1 in code, which is right?

 "tx_gain_lut":[
                {"rf_power": -6, "pa_gain": 0, "pwr_idx":  0},
                {"rf_power": -3, "pa_gain": 0, "pwr_idx":  1},
                {"rf_power":  0, "pa_gain": 0, "pwr_idx":  2},
                {"rf_power":  3, "pa_gain": 1, "pwr_idx":  3},
                {"rf_power":  6, "pa_gain": 1, "pwr_idx":  4},
                {"rf_power": 10, "pa_gain": 1, "pwr_idx":  5},
                {"rf_power": 11, "pa_gain": 1, "pwr_idx":  6},
                {"rf_power": 12, "pa_gain": 2, "pwr_idx":  7},
                {"rf_power": 13, "pa_gain": 1, "pwr_idx":  8},
                {"rf_power": 14, "pa_gain": 2, "pwr_idx":  9},
                {"rf_power": 16, "pa_gain": 2, "pwr_idx": 10},
                {"rf_power": 20, "pa_gain": 3, "pwr_idx": 11},
                {"rf_power": 23, "pa_gain": 3, "pwr_idx": 12},
                {"rf_power": 25, "pa_gain": 3, "pwr_idx": 13},
                {"rf_power": 26, "pa_gain": 3, "pwr_idx": 14},
                {"rf_power": 27, "pa_gain": 3, "pwr_idx": 15}
            ]

piotr-sb, 2020-07-26:
In SX1302 hardware reference design, PA Gain can be only 0 (PA off) or 1 (PA on).

mcoracin, 2020-10-26:
Yes, at least for sx1250 designs, the possible values for pa_gain are 0 or 1.
We will update the global_conf.json.sx1250.CN490 accordingly.
Thanks

closed 2020-10-26 by mcoracin