Lora-net/sx1302_hal #15: SX1261 in corecell design

symovs, 2020-05-31:
Hi

There is a SX1261 transceiver connected to host via SPI. Does the code here handle that transceiver also ? I went through the code but didn't find any corresponding code here. 

Regards
symovs

mcoracin, 2020-06-02:
Hi,
No, it is not supported yet. It will be used later for Listen Before Talk support.
best regards,
Michael

symovs, 2020-06-12:
Can it be used for reception when the base band processor and RF transceivers are transmitting making the gateway full duplex ?

mcoracin, 2020-06-15:
@symovs No, it cannot be used for full-duplex. This radio is behind an RF switch, and will be blind also when the gateway is transmitting. 
Moreover, its RF path is not optimized, so the sensitivity is not good enough for this usage.

mcoracin, 2020-10-26:
Closing as no activity on this.

closed 2020-10-26 by mcoracin