Lora-net/sx1302_hal #16: LEDs in design not lighting up

symovs, 2020-06-12:
The CONFIG_OK, RX and TX LEDs are not lighting up. Is it not being done in code ? If not can you please suggest how it can be done ?

symovs, 2020-06-14:
I saw Rx and Tx LEDs blink for short time yesterday. So I guess CONFIG_OK is the only problem.

symovs, 2020-06-14:
It seems the config ok LED blinks once when the concentrator is started. But thats it. I want it to be permanently ON. I am using the corecell design by the way with a raspberry pi zero host. Pls help.

symovs, 2020-06-14:
I fixed by adding following code in lora_pkt_fwd.c. 

/* Select GPIO_4 to be controlled by HOST */
lgw_reg_w(SX1302_REG_GPIO_GPIO_SEL_6_SELECTION, 0);
/* Configure it as an OUTPUT */
lgw_reg_w(SX1302_REG_GPIO_GPIO_DIR_L_DIRECTION, 0xFF);

/*Set GPIO_6 to high */
lgw_reg_w(SX1302_REG_GPIO_GPIO_OUT_L_OUT_VALUE, 0xFF); 
/* Set GPIO_6 to low */
lgw_reg_w(SX1302_REG_GPIO_GPIO_OUT_L_OUT_VALUE, 0);

when I gave 0x4F or 64 to SX1302_REG_GPIO_GPIO_OUT_L_OUT_VALUE the GPIO4 didn't light up. Am not sure why. It would be helpful if some one can explain how the GPIO registers work.

tonysmith55, 2020-06-22:
I too would like to know how to control the GPIOs 3, 4 and 5 to control the LEDs in the reference design.  I cannot find a document which explains the operation of the GPIO registers.

AloyseTech, 2020-08-17:
Here is a quick sample code for controlling GPIO that works for me : 
```C
void lgw_pin_mode_set(uint8_t pin, bool output)
{
    uint16_t reg_dir, reg_sel;
    int32_t reg_val = 0;

    if (pin < 8)
    {
        reg_dir = SX1302_REG_GPIO_GPIO_DIR_L_DIRECTION;
        reg_sel = SX1302_REG_GPIO_GPIO_SEL_0_SELECTION + pin;
    }
    else if (pin < 12)
    {
        reg_dir = SX1302_REG_GPIO_GPIO_DIR_H_DIRECTION;
        reg_sel = (pin == 8) ? SX1302_REG_GPIO_GPIO_SEL_8_11_GPIO_8_SEL : SX1302_REG_GPIO_GPIO_SEL_8_11_GPIO_11_9_SEL;
    }
    else
        return;

    //set GPIO control mode to HOST
    lgw_reg_w(reg_sel, 0);
    //configure as an output
    lgw_reg_r(reg_dir, &reg_val);
    if(output)
    {
        //set output direction bit
        reg_val |= (1 << (pin % 8));
    }
    else
    {
        //clear output direction bit
        reg_val &= ~(1 << (pin % 8));
    }

    //write modified register value
    lgw_reg_w(reg_dir, reg_val);
}

void lgw_pin_out_write(uint8_t pin, bool high)
{
    uint16_t reg_out;

    // select register depending on pin number
    if (pin < 8)
    {
        reg_out = SX1302_REG_GPIO_GPIO_OUT_L_OUT_VALUE;
    }
    else if (pin < 12)
    {
        reg_out = SX1302_REG_GPIO_GPIO_OUT_H_OUT_VALUE;
    }
    else
        return;

    int32_t reg_val = 0;
    // get output value register
    lgw_reg_r(reg_out, &reg_val);
    if (high)
    {
        // set output value high bit
        reg_val |= (1 << (pin % 8));
    }
    else
    {
        // clear output value high bit
        reg_val &= ~(1 << (pin % 8));
    }

    // Write the register modified value
    lgw_reg_w(reg_out, reg_val);
}
```

shawaj, 2021-10-07:
I believe this has been fixed in updated v3 corecell design and latest codebase

closed 2022-09-30 by smtc-bot