Lora-net/sx1302_hal #17: GPIO control - How does it work? -  Where to find documentation?

tonysmith55, 2020-06-22:
Where can we find details on how to control the registers that control the GPIOs.  

My current interest in the three GPIOs 3, 4 and 5 for the LEDs, but have a wider interest to control all unused GPIOs in the reference design.

Even a short description here outlining the chip's architecture would help point me in the right direction

AloyseTech, 2020-08-17:
Here is a quick sample code for controlling GPIO that works for me : 
```C
#include "loragw_reg.h"

void lgw_pin_mode_set(uint8_t pin, bool output)
{
    uint16_t reg_dir, reg_sel;
    int32_t reg_val = 0;

    if (pin < 8)
    {
        reg_dir = SX1302_REG_GPIO_GPIO_DIR_L_DIRECTION;
        reg_sel = SX1302_REG_GPIO_GPIO_SEL_0_SELECTION + pin;
    }
    else if (pin < 12)
    {
        reg_dir = SX1302_REG_GPIO_GPIO_DIR_H_DIRECTION;
        reg_sel = (pin == 8) ? SX1302_REG_GPIO_GPIO_SEL_8_11_GPIO_8_SEL : SX1302_REG_GPIO_GPIO_SEL_8_11_GPIO_11_9_SEL;
    }
    else
        return;

    //set GPIO control mode to HOST
    lgw_reg_w(reg_sel, 0);
    //configure as an output
    lgw_reg_r(reg_dir, &reg_val);
    if(output)
    {
        //set output direction bit
        reg_val |= (1 << (pin % 8));
    }
    else
    {
        //clear output direction bit
        reg_val &= ~(1 << (pin % 8));
    }

    //write modified register value
    lgw_reg_w(reg_dir, reg_val);
}

void lgw_pin_out_write(uint8_t pin, bool high)
{
    uint16_t reg_out;

    // select register depending on pin number
    if (pin < 8)
    {
        reg_out = SX1302_REG_GPIO_GPIO_OUT_L_OUT_VALUE;
    }
    else if (pin < 12)
    {
        reg_out = SX1302_REG_GPIO_GPIO_OUT_H_OUT_VALUE;
    }
    else
        return;

    int32_t reg_val = 0;
    // get output value register
    lgw_reg_r(reg_out, &reg_val);
    if (high)
    {
        // set output value high bit
        reg_val |= (1 << (pin % 8));
    }
    else
    {
        // clear output value high bit
        reg_val &= ~(1 << (pin % 8));
    }

    // Write the register modified value
    lgw_reg_w(reg_out, reg_val);
}
```

tonysmith55, 2020-08-17:
Thanks @AloyseTech that will work for me.  I had worked out some of this but no where near the sophistication of your code.  Had not looked beyond bit 7 and had all GPIOs as inputs or outputs not separately controlled as you have done.

closed 2022-09-30 by smtc-bot