Lora-net/sx1302_hal #20: cross compiling

RAhul2413, 2020-09-16:
Hello Every one

I am new hear and little bit of coding knoweledge .now i want to run packate forwarder on stm32mp157 (STM microprocessor)
for that i am using yocto SDK.now i want to cross compile this code but it shows error tinymt32.h  not found.any one know how to add and where to add this file.or which package i need to install.

mcoracin, 2020-09-16:
Hello,
The tinymt32 library is only there for debug/test helpers. You can safely remove the code using it and don't bother with this for your porting.
Best regards,
Michael

RAhul2413, 2020-09-16:
thanks for reply,

i understand what r u say about tinymt.but how to solve this error.r u telling me to remove the tinymt from Makefile code.or something else.

mcoracin, 2020-09-16:
yes, either you find a way to have it compiled ok for st32, I don't see any reason why you can't do that, but you can also just remove it from the makefile and of course comment out the code using this library to avoid compilation error.

RAhul2413, 2020-09-16:
thanks for reply,

i did the same u told me but after that parson.h not found error appears.just tell me one thing when i cross compile it where it search for the header files.when i normally compile it.it compiled perfectly. when i cross compile it than shows error.

RAhul2413, 2020-09-16:
and tell me one thing  should i  first execute make install and make install_conf .commands than after i should cross compile it or its fine to first cross compile.

mcoracin, 2020-09-16:
make install if for copying the compiled binaries and configuration files to the host linux filesystem. So you need to cross compile first, then install

RAhul2413, 2020-09-17:
thanks for reply .

RAhul2413, 2020-09-17:
now i am able to cross compile the code and finally code is running on my stm32mp1(MPU).

closed 2020-09-17 by RAhul2413