Lora-net/sx1302_hal #23: Packet drop in Sx1302

vikasrohal, 2020-10-20:
After multiple testing I have observed that there are some LoRa packets drops in gateway designed from sx1302 , in comparison with sx1301 for which I have also attached some screenshots. There are min of 5% packets drops in sx1302 . I hope that someone have faced similar issue can help me in some way. thanks for help in advance. 

[image]
[image]

rony-vargas, 2020-10-22:
I have experience similar issue, I have a RAK2245 and RAK2287 (SX1301 vs SX1301) and the first captures way more packets than the second. I am looking various forks from this repo to see if there is any improvement that I can implement locally

rony-vargas, 2020-10-23:
Hi @vikasrohal , what packet forwarder are you using? could you print the output when the concentrator SX1302 starts? I am interested to see the `rssi_offset` in my case I suspect it could be part of the problem, I am using the `basicstation` as packet forwarder and it is not honoring the `rssi_offset` of -215.4 set in the `station.conf`, configurations may if you're using a different paket forwarder, but in some place of a you should define a rssi_offset based on your board's desing.
Since I believe it is more related the specific forwarder I am using I have opened an issue in that repository:
[https://github.com/lorabasics/basicstation/issues/80](url)

rony-vargas, 2020-10-26:
@vikasrohal I found my issue to be in the node side, it was using a wrong preamble in the code, LoRaWAN establishes that it needs to be 8 Symbols for preamble and in my case the node had a code bug that reported more than that. It made the Gateway to discard that message. After I fixed the Preamble length in the node's code, I had no more packet drop.
Apparently the `rssi_offset` has nothing to do with this issue, but still looking forward the TheThingsStack to avoid pushing this for the LNS protocol and avoid hardware issue.

mcoracin, 2020-12-18:
Closing this issue, as it was not related to this project

closed 2020-12-18 by mcoracin