Lora-net/sx1302_hal #28: Why mix code from different reference designs?

wong-hao, 2020-12-23:
For example, the 1.0.5 version code can be used normally on my device, but the latest 2.0.1 can't work because there are more non-existent devices such as sx1261 and ADC ad338r. If I want to use the code, I can only check the previous code version.

Even if I want to use the latest version of the code by commenting out the code that calls these devices, it is very hard.

mcoracin, 2020-12-23:
If you had a device which was working with the v1.0.5, it can work with the v2.0.1 too.
The AD338R is only enabled if full-duplex is enabled. If full-duplex is disabled, it won't try to initialize it, so no problem.
Same for sx1261, if you don't have this extra radio on your design, just leave lbt and spectral scan disabled, and no problem either.

Which gateway design are you using ? What problem do you see when running the v2.0.1 ?

wong-hao, 2020-12-23:
I am using the design [Semtech SX1302CxxxGW1](https://www.semtech.com/products/wireless-rf/lora-gateways/sx1302cxxxgw1)

An error occurred when I started the packet forwarder: 
[image]

The configuration file I'm using is almost exactly the same as the default file global_conf.json.sx1250.CN490 provided, and I don't think it's possible to make it work by turning off some features rather than adjusting the code:
[image]

mcoracin, 2020-12-23:
Ok, we'll have a look into this when we'll be back from christmas holidays, beginning of january.
This board has not been tested with this release, so there is maybe an issue.
But we'll fix it.

wong-hao, 2020-12-27:
My fault, because there is no temperature sensor (I2C device) in my design, I need to comment out some I2C-related code. Version 2.0.1 needs more comments so I made a mistake before.

It would be great if there is an option like full-duplex and lbt to choose whether to enable the temperature sensor (I2C). Of course, I just make an opinion.

mcoracin, 2021-01-05:
Hello,
Do you mean that the radio setup error you were getting was due to I2C ?

wong-hao, 2021-01-05:
> Hello,
> Do you mean that the radio setup error you were getting was due to I2C ?

The main reason is that I didn't have the problem of I2C device. Later, I commented out the relevant code in the [file](https://github.com/Lora-net/sx1302_hal/blob/master/libloragw/src/loragw_hal.c) and it can run successfully

mcoracin, 2021-01-05:
But I still don't understand why having no temperature sensor lead you to get the SX1250 setup error you shown in your comment https://github.com/Lora-net/sx1302_hal/issues/28#issuecomment-750330581

If no temperature sensor is found, you should get the corresponding error from here:
https://github.com/Lora-net/sx1302_hal/blob/05416800cd3f3681db49201fe4335835ef91c1a2/libloragw/src/loragw_hal.c#L1110

Could you explain what you changed exactly ?

wong-hao, 2021-01-06:
Well, this error is caused by the hardware manufacturer I bought. 

What I mean is that the gateway can work normally means that after reporting this error once, there is no such error for the second time. 

The gateway can't work means it always reporting this error.

mcoracin, 2021-01-06:
The Corecell reference design includes a temperature sensor in order to compensate the temperature impact on the reported RSSI.
So for now, we expect the gateway to have a temperature sensor, in order to be precise with the RSSI reported.

closed 2021-01-06 by mcoracin