Lora-net/sx1302_hal #32: Frequency values being rounded to 100 KHz resolution.

reissjason, 2021-03-01:
When the gateway is used for IN865 the frequencies reported by the packet forwarder are being changed from the configured channels. When the reported channel is used for the Rx1 window the end-device may miss the packet as it may be 25KHz from the actual channel.

Demonstrating with 915 hardware shows the frequency and foff. Should the foff be used to adjust the Rx Frequency?

Expecting to receive a packet at 915.385 MHz

INFO: radio 0 enabled (type SX1250), center frequency 915200000, RSSI offset -196.000000, tx enabled 1, single input mode 0
INFO: radio 1 enabled (type SX1250), center frequency 915385000, RSSI offset -196.000000, tx enabled 0, single input mode 0
INFO: no configuration for LoRa multi-SF spreading factors enabling
INFO: Lora multi-SF channel 0>  radio 1, IF -137500 Hz, 125 kHz bw, SF 5 to 12
INFO: Lora multi-SF channel 1>  radio 1, IF 202500 Hz, 125 kHz bw, SF 5 to 12
INFO: Lora multi-SF channel 2>  radio 1, IF 32500 Hz, 125 kHz bw, SF 5 to 12
INFO: Lora multi-SF channel 3>  radio 0, IF -400000 Hz, 125 kHz bw, SF 5 to 12
INFO: Lora multi-SF channel 4>  radio 0, IF -200000 Hz, 125 kHz bw, SF 5 to 12
INFO: Lora multi-SF channel 5>  radio 0, IF 0 Hz, 125 kHz bw, SF 5 to 12
INFO: Lora multi-SF channel 6>  radio 0, IF 200000 Hz, 125 kHz bw, SF 5 to 12
INFO: Lora multi-SF channel 7>  radio 0, IF 400000 Hz, 125 kHz bw, SF 5 to 12

JSON up: {"rxpk":[{"jver":1,"tmst":4875057,"chan":6,"rfch":0,"freq":915.400000,"mid": 9,"stat":1,"modu":"LORA","datr":"SF10BW125","codr":"4/5","rssis":3,"lsnr":12.2,"foff":15044,"rssi":3,"size":18,"data":"gDQSNBKBCAANAesmY6ZqeFfo"}]}

mcoracin, 2021-03-01:
Hello,
Before looking further, you say that the packet is being emitted at 915.385MHz, right ?
But, on which gateway channel would you expect to receive it ? I don't see a channel configured for this frequency, right ?

reissjason, 2021-03-01:
You're right. The radio offsets are mismatched to what I was expecting. 
After changing the assigned radios I am able to Rx at 915.385 MHz

closed 2021-03-12 by mcoracin