Lora-net/sx1302_hal #37: How to know SX1302 Receiver state (it is reeving lora packets or not)

mightChamp, 2021-03-30:
Hello,

Power Fluctuation is a test condition where high voltages switching took place outside of system condition to get Hardware failure rate. In this, system have to recover to its state again if anything fails, As gateway is places at higher altitude and also there may be lighting that also can cause this type of issue.
So we have to make it recoverable in this state. we can’t make system to stuck somewhere.

When there is power fluctuations, SX1302 stop receiving and didn’t receive any data till sx1302 reset.

But for resetting, I need to know the state of sx1302, if it is receiving or not problematically.

So, How do I know SX1302 is in receving mode or not?

Thank You

nilesh-dryad, 2021-05-06:
Hello mightChamp,

I am also facing the same issue. Due some random spike from my controller's pin (used to reset sx1302), sx1302 stops receiving data. There is no other option for us except to reset the sx1302. 

As of now we did two things, we keep on monitoring receiving data, when host controller does not receive packet for more than 3 minutes, we reset the sx1302 from host controller. This is just workaround.

One more thing, we can just disconnect reset pin after resetting sx1302. This is not an ideal solution but for testing it will work for now. As you asked there is must be some receiver state option based on that we can decide further action.

Do you have any update?

mightChamp, 2021-05-28:
lgw_get_instcnt() will remains constant if receiver stop receiving I used this hack to know the state of receiver, if its value not change then i reset sx1302.

nilesh-dryad, 2021-05-28:
Thanks for the suggestion. Let me try that.

mcoracin, 2021-06-04:
There is no simple status register to read to get this information.
Reading the counter is an idea yes.
Also, it is possible to get the LoRa packet detection status by using the sx1302_arb_get_debug_stats_detect() function on all channels.
The sx1302_arb_set_debug_stats() function should be called before with SF7 for example.
As there are regular false detection of SF7 packets, you should see the detect counter increasing very frequently, even if there is no actual traffic.
If this counter does not increase for a certain amount of time (to be determined), it could be due to the fact that your sx1302 is hanged...
But it requires some experimentation on your side to find the proper threshold.

closed 2021-06-19 by mightChamp