Lora-net/sx1302_hal #54: No documentation to calculate RSSI in dBm for SX1302 + SX1250

nilesh-dryad, 2021-08-19:
I am using Semtech LoRa® Corecell reference design V1.0 (which has SX1302 and SX1250) and testing Receive and Transmit examples. While receiving packets, packet information are printed on console which has RSSI channel and RSSI signal value but those values are in dB.

Is there any equation to convert those values in dBm? I tried to find the information in source code and documents from Semtech website but all in vain.

nilesh-dryad, 2021-11-12:
@kmuster-semtech
Thank you for your reply.
 How can I find that proper RSSI offset? is it fixed -215.4 for SX1302+ SX1250?

kmuster-semtech, 2021-12-21:
How can I find that proper RSSI offset?
The RSSI Offset value is found by calibration, shown in the CoreCell V1 Performance Report I linked earlier. The default value should be proper for any Corecell V1, but you may perform the calibration yourself and update the global_conf.json file of the packet forwarder.

closed 2022-09-30 by smtc-bot