Lora-net/sx1302_hal #61: Failed to set SX1250_0 in STANDBY_RC mode

shawaj, 2021-10-08:
I am getting this error:

```
root@f986d0ed3f2c:/opt/iotloragateway/packet_forwarder# sx1302/util_chip_id/chip_id -d /dev/spidev1.2
CoreCell reset through GPIO22...
Opening SPI communication interface
Note: chip version is 0x12 (v1.2)
ERROR: Failed to set SX1250_0 in STANDBY_RC mode
ERROR: failed to setup radio 0
ERROR: failed to start the gateway
```

And the concentrator gets very hot, very quickly. 

Any idea of the issue?

cstratton, 2021-12-01:
>And the concentrator gets very hot, very quickly

That would indicate a hardware issue.  Either mistaken connection of the concentrator card to the host system (which card, in which system, connected how?) or else past permanent damage from miswiring, ESD, etc.

gswebspace, 2021-12-21:
Noticed `SX1250_0 in STANDBY_RC mod` in my logs too, but not sure if it is due to the same reason. Opened #67 to track it.

wong-hao, 2021-12-22:
> I have seen that error when there is an issue with the hardware setup. For general troubleshooting, make sure the particular concentrator board supports SPI and not USB communication and that the SX1250 transceiver is receiving 3.3V supply power. Check that you are able to run "sx1302/bin/test_loragw_com -d /dev/spidev1.2" and "sx1302/bin/test_loragw_com_1250 -d /dev/spidev1.2" and receive a stream of outputs saying "no error." After checking that, can you also share the concentrator PCB number and any mounted power jumpers?

I'm sure it's not a personal problem because people who use different brands of hardware and different versions of library will all encounter this problem. Please refer to [this issue](https://github.com/Lora-net/sx1302_hal/issues/10)

cstratton, 2021-12-22:
>I'm sure it's not a personal problem because people who use different brands of hardware and different versions of library will all encounter this problem.

No, while the error message is commonly seen, the cause is very much a "personal" one unique to a given setup or hardware problem.

Most commonly, that would be not having scripts configured to use the GPIO that is actually wired to the concentrator reset in your particular configuration.  Other causes have included the one earlier in this thread where someone has a damaged concentrator chip, and one at your link where someone is trying to do an unsupported "hot change" of configuration without reset.

LittleFool1022, 2022-03-04:
I had the same problem, on Raspberry Pi 3B and SX1302 gateway module, the problem was failed to set sx1250_0 STANDBY_RC mode. However, I can get the chip version by pulling the CSN of the SX1302 and the SPI chip selection control pin by pulling it high and then pulling it low, but the above problems are still impossible.

ahmed560, 2022-05-24:
but the problem isn't resolved

closed 2022-09-30 by smtc-bot

ABHITHLALC, 2023-07-14:
Did anyone solved this problem? I am having the same problem. I am using  raspberry pi 3b+ with sx1302 lorawan hat

jtbuffmire, 2025-08-20:
This was killing me. Turned out to be a race condition between my gateway-service and packet-forwarder on the RAK 2287 and r-pi4. The gateway-rs service took ~13s to load. RAK's was beating it every time. Adding a 15s ``STARTUP_DELAY`` to the RAK UDP packet-forwarder service in the docker-compose solved it for me. https://github.com/jtbuffmire/helium-gateway-balena