Lora-net/sx1302_hal #66: AGC fw read/write check failed

gswebspace, 2021-12-20:
The packet forwarder refuses to start with the following error -

```
sx1302_hal  *** Packet Forwarder ***
 sx1302_hal  Version: 2.1.0
 sx1302_hal  *** SX1302 HAL library version info ***
 sx1302_hal  Version: 2.1.0;
 sx1302_hal  ***
 sx1302_hal  INFO: Little endian host
 sx1302_hal  INFO: found configuration file global_conf_US915.json, parsing it
 sx1302_hal  INFO: global_conf_US915.json does contain a JSON object named SX130x_conf, parsing SX1302 parameters
 sx1302_hal  INFO: com_type SPI, com_path /dev/spidev0.0, lorawan_public 1, clksrc 0, full_duplex 0
 sx1302_hal  INFO: antenna_gain 0 dBi
 sx1302_hal  INFO: Configuring legacy timestamp
 sx1302_hal  INFO: Configuring Tx Gain LUT for rf_chain 0 with 16 indexes for sx1250
 sx1302_hal  INFO: radio 0 enabled (type SX1250), center frequency 904300000, RSSI offset -215.399994, tx enabled 1, single input mode 0
 sx1302_hal  INFO: radio 1 enabled (type SX1250), center frequency 905000000, RSSI offset -215.399994, tx enabled 0, single input mode 0
 sx1302_hal  INFO: no configuration for LoRa multi-SF spreading factors enabling
 sx1302_hal  INFO: Lora multi-SF channel 0>  radio 0, IF -400000 Hz, 125 kHz bw, SF 5 to 12
 sx1302_hal  INFO: Lora multi-SF channel 1>  radio 0, IF -200000 Hz, 125 kHz bw, SF 5 to 12
 sx1302_hal  INFO: Lora multi-SF channel 2>  radio 0, IF 0 Hz, 125 kHz bw, SF 5 to 12
 sx1302_hal  INFO: Lora multi-SF channel 3>  radio 0, IF 200000 Hz, 125 kHz bw, SF 5 to 12
 sx1302_hal  INFO: Lora multi-SF channel 4>  radio 1, IF -300000 Hz, 125 kHz bw, SF 5 to 12
 sx1302_hal  INFO: Lora multi-SF channel 5>  radio 1, IF -100000 Hz, 125 kHz bw, SF 5 to 12
 sx1302_hal  INFO: Lora multi-SF channel 6>  radio 1, IF 100000 Hz, 125 kHz bw, SF 5 to 12
 sx1302_hal  INFO: Lora multi-SF channel 7>  radio 1, IF 300000 Hz, 125 kHz bw, SF 5 to 12
 sx1302_hal  INFO: Lora std channel> radio 0, IF 300000 Hz, 500000 Hz bw, SF 8, Explicit header
 sx1302_hal  INFO: FSK channel 8 disabled
 sx1302_hal  INFO: global_conf_US915.json does contain a JSON object named gateway_conf, parsing gateway parameters
 sx1302_hal  INFO: gateway MAC address is configured to AA555A0000000000
 sx1302_hal  INFO: server hostname or IP address is configured to "miner"
 sx1302_hal  INFO: upstream port is configured to "1680"
 sx1302_hal  INFO: downstream port is configured to "1680"
 sx1302_hal  INFO: downstream keep-alive interval is configured to 10 seconds
 sx1302_hal  INFO: statistics display interval is configured to 30 seconds
 sx1302_hal  INFO: upstream PUSH_DATA time-out is configured to 100 ms
 sx1302_hal  INFO: packets received with a valid CRC will be forwarded
 sx1302_hal  INFO: packets received with a CRC error will NOT be forwarded
 sx1302_hal  INFO: packets received with no CRC will NOT be forwarded
 sx1302_hal  INFO: global_conf_US915.json does not contain a JSON object named debug_conf
 sx1302_hal  INFO: no debug configuration
 sx1302_hal  CoreCell reset through GPIO23...
 sx1302_hal  SX1261 reset through GPIO23...
 sx1302_hal  CoreCell power enable through GPIO18...
 sx1302_hal  CoreCell ADC reset through GPIO13...
 sx1302_hal  Opening SPI communication interface
 sx1302_hal  Note: chip version is 0x00 (v0.0)
 sx1302_hal  INFO: using legacy timestamp
 sx1302_hal  INFO: LoRa Service modem: configuring preamble size to 8 symbols
 sx1302_hal  ERROR: AGC fw read/write check failed
 sx1302_hal  ERROR: failed to load AGC firmware for sx1250
 sx1302_hal  ERROR: [main] failed to start the concentrator
```

Note the last 3 lines -

```
 sx1302_hal  ERROR: AGC fw read/write check failed
 sx1302_hal  ERROR: failed to load AGC firmware for sx1250
 sx1302_hal  ERROR: [main] failed to start the concentrator
```

What does the error mean ? and can we debug/fix this ?

louischuang, 2022-05-02:
I have the same issue too.

closed 2022-09-30 by smtc-bot

RoyceThomasIype, 2023-05-31:
@gswebspace , @louischuang Did you fix the issue? If so, can you please share the fix? I'm also facing the same issue.

gswebspace, 2023-06-04:
@RoyceThomasIype I don't think I got a solution for this.

But, looking back and considering that the version shows up as `0x00 (v0.0)` I suspect this may be related to #67 and the concentrator has not been reset properly. The subsequent error thrown by the AGC may be misleading.

I'll suggest to ensure the reset happens properly. See this comment that fixed the reset issue for me : https://github.com/Lora-net/sx1302_hal/issues/67#issuecomment-1024505406