Lora-net/sx1302_hal #67: Failed to set SX1250_0 in STANDBY_RC mode

gswebspace, 2021-12-21:
Note: The device is using the [RAK2287 Pi HAT](https://docs.rakwireless.com/Product-Categories/WisHat/RAK2287-Pi-HAT/Datasheet/#overview), which uses [RAK2287 concentrator](https://docs.rakwireless.com/Product-Categories/WisLink/RAK2287/Datasheet/) with one SX1302 chip and two SX1250.

The packet forwarder (latest version) fails during startup with the following logs -
```
.
.
CoreCell reset through GPIO17...
SX1261 reset through GPIO17...
CoreCell power enable through GPIO18...
CoreCell ADC reset through GPIO13...
Opening SPI communication interface
Note: chip version is 0x05 (v0.5)
ERROR: Failed to set SX1250_0 in STANDBY_RC mode
ERROR: failed to setup radio 0
ERROR: [main] failed to start the concentrator
```

Tried using the chip_id tool -
```
root@bf8cb7d6a21e:/opt/sx1302_hal/bin# ./chip_id 
CoreCell reset through GPIO17...
SX1261 reset through GPIO17...
CoreCell power enable through GPIO18...
CoreCell ADC reset through GPIO13...
Opening SPI communication interface
Note: chip version is 0x05 (v0.5)
ERROR: Failed to set SX1250_0 in STANDBY_RC mode
ERROR: failed to setup radio 0
ERROR: failed to start the gateway
```

Interestingly power cycling (power off and on) helps. Logs after power cycle -
```
root@bf8cb7d6a21e:/opt/sx1302_hal/bin# ./chip_id 
CoreCell reset through GPIO17...
SX1261 reset through GPIO17...
CoreCell power enable through GPIO18...
CoreCell ADC reset through GPIO13...
Opening SPI communication interface
Note: chip version is 0x10 (v1.0)
INFO: using legacy timestamp
ARB: dual demodulation disabled for all SF
INFO: found temperature sensor on port 0x39

INFO: concentrator EUI: 0x0016c001ff19dccd

Closing SPI communication interface
CoreCell reset through GPIO17...
SX1261 reset through GPIO17...
CoreCell power enable through GPIO18...
CoreCell ADC reset through GPIO13...
```

But after the above command, if it is ran again, the problems shows up again -
```
root@bf8cb7d6a21e:/opt/sx1302_hal/bin# ./chip_id 
CoreCell reset through GPIO17...
SX1261 reset through GPIO17...
CoreCell power enable through GPIO18...
CoreCell ADC reset through GPIO13...
Opening SPI communication interface
Note: chip version is 0x10 (v1.0)
ERROR: Failed to set SX1250_0 in STANDBY_RC mode
ERROR: failed to setup radio 0
ERROR: failed to start the gateway
```

Note: Chip version goes from `v0.5` to `v1.0` after a power cycle, which is probably not normal.

Not sure how to fix/debug this ? Any help is appreciated. Thanks!

kmuster-semtech, 2021-12-21:
The RAK2287 has its own firmware designed to work with the board described [here](https://docs.rakwireless.com/Product-Categories/WisLink/RAK2287/Quickstart/#prerequisites). When using the sx1302_hal with the CoreCell board (PCB_E539V03A, [user guide here](https://semtech.my.salesforce.com/sfc/p/#E0000000JelG/a/2R000000Hokn/i4xzbsLU2V6V92uDISoUfd3brd4MOpa0kyPyC_FPKXo)), additional parameters are needed for the USB version with MCU mounted. Since the RAK2287 concentrator also has a MCU, try the following command instead:
./chip_id -u -d /dev/ttyACM0

mmareks, 2022-01-27:
I have the same problem with lora concentrator module RAK2287.   Sometimes i t work after power cycle, also noticed switching version from 0.5 to 1.0. Tried RAK's firmware for RAK2287 as suggested but got the same problem. Also tried test_loragw_com and test_loragw_com_sx1250 but it READS only zero. 

`
pi@rak-gateway:/usr/local/rak/lora/rak2287_spi/lora_gateway $ ./test_loragw_com_sx1250 -d /dev/spidev0.0
CoreCell reset through GPIO17...
Opening SPI communication interface
Note: chip version is 0x00 (v0.0)
Radio0: get_status: 0x00
Radio1: get_status: 0x00
Cycle 0 > error during the buffer comparison
Written value: 67C66973
Read value:    00000000
`
Any suggestion to debug ?

gswebspace, 2022-01-27:
I can confirm similar results.

@kmuster-semtech Installed the latest [RAK firmware](https://docs.rakwireless.com/Product-Categories/WisLink/RAK2287/Quickstart/#prerequisites) as suggested. Same error.

## RAK Firmware
```
root@rak-gateway:/opt/ttn-gateway/packet_forwarder/lora_pkt_fwd# ./start.sh 
./start.sh: line 6: echo: write error: Device or resource busy
*** Packet Forwarder ***
Version: 2.0.1
*** SX1302 HAL library version info ***
Version: 2.0.1;
***
INFO: Little endian host
INFO: found configuration file global_conf.json, parsing it
INFO: global_conf.json does contain a JSON object named SX130x_conf, parsing SX1302 parameters
INFO: com_type SPI, com_path /dev/spidev0.0, lorawan_public 1, clksrc 0, full_duplex 0
INFO: antenna_gain 0 dBi
INFO: Configuring legacy timestamp
INFO: no configuration for SX1261
INFO: Configuring Tx Gain LUT for rf_chain 0 with 16 indexes for sx1250
INFO: radio 0 enabled (type SX1250), center frequency 904300000, RSSI offset -215.399994, tx enabled 1, single input mode 0
INFO: radio 1 enabled (type SX1250), center frequency 905000000, RSSI offset -215.399994, tx enabled 0, single input mode 0
INFO: Lora multi-SF channel 0>  radio 0, IF -400000 Hz, 125 kHz bw, SF 5 to 12
INFO: Lora multi-SF channel 1>  radio 0, IF -200000 Hz, 125 kHz bw, SF 5 to 12
INFO: Lora multi-SF channel 2>  radio 0, IF 0 Hz, 125 kHz bw, SF 5 to 12
INFO: Lora multi-SF channel 3>  radio 0, IF 200000 Hz, 125 kHz bw, SF 5 to 12
INFO: Lora multi-SF channel 4>  radio 1, IF -300000 Hz, 125 kHz bw, SF 5 to 12
INFO: Lora multi-SF channel 5>  radio 1, IF -100000 Hz, 125 kHz bw, SF 5 to 12
INFO: Lora multi-SF channel 6>  radio 1, IF 100000 Hz, 125 kHz bw, SF 5 to 12
INFO: Lora multi-SF channel 7>  radio 1, IF 300000 Hz, 125 kHz bw, SF 5 to 12
INFO: Lora std channel> radio 0, IF 300000 Hz, 500000 Hz bw, SF 8, Explicit header
INFO: FSK channel 8 disabled
INFO: global_conf.json does contain a JSON object named gateway_conf, parsing gateway parameters
INFO: gateway MAC address is configured to E45F01FFFE2069FB
INFO: server hostname or IP address is configured to "nam1.cloud.thethings.network"
INFO: upstream port is configured to "1700"
INFO: downstream port is configured to "1700"
INFO: downstream keep-alive interval is configured to 10 seconds
INFO: statistics display interval is configured to 30 seconds
INFO: upstream PUSH_DATA time-out is configured to 100 ms
INFO: packets received with a valid CRC will be forwarded
INFO: packets received with a CRC error will NOT be forwarded
INFO: packets received with no CRC will NOT be forwarded
INFO: GPS serial port path is configured to "/dev/ttyAMA0"
INFO: Reference latitude is configured to 0.000000 deg
INFO: Reference longitude is configured to 0.000000 deg
INFO: Reference altitude is configured to 0 meters
INFO: Beaconing period is configured to 0 seconds
INFO: Beaconing signal will be emitted at 923300000 Hz
INFO: Beaconing channel number is set to 8
INFO: Beaconing channel frequency step is set to 600000Hz
INFO: Beaconing datarate is set to SF12
INFO: Beaconing modulation bandwidth is set to 500000Hz
INFO: Beaconing TX power is set to 27dBm
INFO: global_conf.json does contain a JSON object named debug_conf, parsing debug parameters
INFO: got 2 debug reference payload
INFO: reference payload ID 0 is 0xCAFE1234
INFO: reference payload ID 1 is 0xCAFE2345
INFO: setting debug log file name to loragw_hal.log
INFO: found configuration file local_conf.json, parsing it
INFO: local_conf.json does contain a JSON object named gateway_conf, parsing gateway parameters
INFO: gateway MAC address is configured to E45F01FFFE2069FB
INFO: packets received with a valid CRC will be forwarded
INFO: packets received with a CRC error will NOT be forwarded
INFO: packets received with no CRC will NOT be forwarded
INFO: [main] TTY port /dev/ttyAMA0 open for GPS synchronization
CoreCell reset through GPIO17...
Opening SPI communication interface
Note: chip version is 0x05 (v0.5)
[... 139 lines cut ...]
 05  05  05  05  05  05  05  05  05  05  05  05  05  05  05  05 
 05  05  05  05  05  05  05  05  05  05  05  05  05  05  05  05 
 05  05  05  05  05  05  05  05  05  05  05  05  05  05  05  05 
 05  05  05  05  05  05  05  05  05  05  05  05  05  05  05  05 
 05  05  05  05  05  05  05  05  05  05  05  05  05  05  05  05 
 05  05  05  05  05  05  05  05  05  05  05  05  05  05  05  05 
 05  05  05  05  05  05  05  05  05  05  05  05  05  05  05  05 
 05  05  05  05  05  05  05  05  05  05  05  05  05  05  05  05 
 05  05  05  05  05  05  05  05  05  05  05  05  05  05  05  05 
 05  05  05  05  05  05  05  05  05  05  05  05  05  05  05  05 
 05  05  05  05  05  05  05  05  05  05  05  05  05  05  05  05 
 05  05  05  05  05  05  05  05  05  05  05  05  05  05  05  05 
 05  05  05  05  05  05  05  05  05  05  05  05  05  05  05  05 
 05  05  05  05  05  05  05  05  05  05  05  05  05  05  05  05 
 05  05  05  05  05  05  05  05  05  05  05  05  05  05  05  05 
 05  05  05  05  05  05  05 
Closing SPI communication interface
End of test for loragw_com.c
```

Note: The Read Values are a fixed value.

Also in `/opt/ttn-gateway/packet_forwarder/lora_pkt_fwd`, the reset pin is set as `SX1302_RESET_PIN=17`.

Is there a way to read any error codes from the sx1302 chip to understand what is the exact error ? Any specific register that could be read ?

mareksbb, 2022-01-28:
```SX1302_RESET_PIN=25```
seems to solve the problem

gswebspace, 2022-01-28:
@mareksbb Thanks for sharing ! It certainly is helping. The packet forwarder seems to be working fine now. 

GPIO 25 seems to be connected to the GPS modules's RESET pin (on the RAK2287 Pi HAT).
I have modified the reset_lgw.sh to do a GPS reset and then the sx1302 reset.

Starting longevity test to check stability.

The original cause of the issue is still unknown and needs investigation.

ahmed560, 2022-06-01:
this library is configured for raspberry pi os 32 bits. it's not compatible with 64 bits architecture

gswebspace, 2022-06-03:
@ahmed560 I have tested sx1302_hal with 64 bit os on Rpi4B. It is compatible as far as I can see it.

ahmed560, 2022-06-03:
So, I did a cross-compilation  with aarch64-linux-gnu  toolchain for this
project and i copied libraries and binary files in raspberry card  the is
worked but the RX test show this problem

gswebspace, 2022-06-07:
@ahmed560 If you see the error `Failed to set SX1250_0 in STANDBY_RC mode` then so far the only solution is to try using GPIO25 , see : https://github.com/Lora-net/sx1302_hal/issues/67#issuecomment-1024326428

Exact cause of the issue is not yet known.

closed 2022-09-30 by smtc-bot

chaudhariatul, 2022-11-27:
I'm fac

mcoracin, 2022-11-28:
This issue is generally due to a bad SPI connection/configuration.
You need to have test_loragw_com running with no error first.

chaudhariatul, 2022-11-28:
@mcoracin Initially I had the same assumption that the SPI connection is bad.

However, I am able to use this setup with https://github.com/xoseperez/basicstation and able to connect it to ThingStack. 

I've reached out to the rak forum for help too.

mcoracin, 2022-11-29:
@chaudhariatul if the SPI is ok, I guess the issue would be around the reset_lgw.sh script, maybe the GPIO used on the RAK gateway are not the same ?
I guess the RAK support should be able to help for that.

```
CoreCell reset through GPIO25...
SX1261 reset through GPIO25...
CoreCell power enable through GPIO18...
CoreCell ADC reset through GPIO13...
```

mcoracin, 2022-11-29:
I'm a bit surprised by the "chip version is 0x05 (v0.5)", I don't think this version exists.
Could you try with SPI speed at 2MHz instead of 8MHz ?

chaudhariatul, 2022-11-29:
Changing the speed to 2Mhz still shows `chip version is 0x05 (v0.5)`

```
basicstation    | 2022-11-29 17:01:03.910 [S00:INFO] Station device: spi:/dev/spidev0.0 (PPS capture enabled)
basicstation    | 2022-11-29 17:01:03.910 [S00:INFO] [lgw_com_open:84] Opening SPI communication interface
basicstation    | 2022-11-29 17:01:03.910 [S00:INFO] [lgw_spi_open:88] Setting SPI speed to 2000000
basicstation    | 2022-11-29 17:01:03.910 [S00:INFO] [lgw_connect:1192] chip version is 0x05 (v0.5)
basicstation    | 2022-11-29 17:01:03.787 [RAL:INFO] Region plan hwspec 'sx1301/1' mapped to 1 slaves 'sx1301/1'
basicstation    | 2022-11-29 17:01:03.787 [RAL:INFO] Master sending 645 bytes of JSON sx1301conf to slave (0)
basicstation    | 2022-11-29 17:01:03.788 [S2E:INFO] Configuring for region: US915 -- 923.0MHz..928.0MHz
basicstation    | 2022-11-29 17:01:03.788 [S2E:INFO]   DR0  SF10/BW125
basicstation    | 2022-11-29 17:01:03.788 [S2E:INFO]   DR1  SF9/BW125
basicstation    | 2022-11-29 17:01:03.788 [S2E:INFO]   DR2  SF8/BW125
basicstation    | 2022-11-29 17:01:03.788 [S2E:INFO]   DR3  SF7/BW125
basicstation    | 2022-11-29 17:01:03.788 [S2E:INFO]   DR4  SF8/BW500
basicstation    | 2022-11-29 17:01:03.788 [S2E:INFO]   DR5  FSK
basicstation    | 2022-11-29 17:01:03.788 [S2E:INFO]   DR6  FSK
basicstation    | 2022-11-29 17:01:03.788 [S2E:INFO]   DR7  FSK
basicstation    | 2022-11-29 17:01:03.788 [S2E:INFO]   DR8  SF12/BW500
basicstation    | 2022-11-29 17:01:03.788 [S2E:INFO]   DR9  SF11/BW500
basicstation    | 2022-11-29 17:01:03.788 [S2E:INFO]   DR10 SF10/BW500
basicstation    | 2022-11-29 17:01:03.788 [S2E:INFO]   DR11 SF9/BW500
basicstation    | 2022-11-29 17:01:03.788 [S2E:INFO]   DR12 SF8/BW500
basicstation    | 2022-11-29 17:01:03.788 [S2E:INFO]   DR13 SF7/BW500
basicstation    | 2022-11-29 17:01:03.788 [S2E:INFO]   DR14 FSK
basicstation    | 2022-11-29 17:01:03.788 [S2E:INFO]   DR15 FSK
basicstation    | 2022-11-29 17:01:03.788 [S2E:INFO]   TX power: 26.0 dBm EIRP
basicstation    | 2022-11-29 17:01:03.788 [S2E:INFO]   JoinEui list: 0 entries
basicstation    | 2022-11-29 17:01:03.788 [S2E:INFO]   NetID filter: FFFFFFFF-FFFFFFFF-FFFFFFFF-FFFFFFFF
basicstation    | 2022-11-29 17:01:03.788 [S2E:INFO]   Dev/test settings: nocca=1 nodc=1 nodwell=1
basicstation    | ERROR: Failed to set SX1250_0 in STANDBY_RC mode
basicstation    | 2022-11-29 17:01:05.512 [S00:ERRO] [lgw_start:908] failed to setup radio 0
basicstation    | 2022-11-29 17:01:05.512 [S00:ERRO] Concentrator start failed: lgw_start
basicstation    | 2022-11-29 17:01:05.512 [S00:CRIT] Slave radio start up failed with status 0x08
basicstation    | 2022-11-29 17:01:05.478 [RAL:ERRO] Slave (0) - EOF
basicstation    | 2022-11-29 17:01:05.478 [any:CRIT] Slave pid=34 idx=0: Fatal exit
```

mcoracin, 2022-11-30:
Which gateway hardware are you using exactly, and which version of the HAL ?
The version should be 1.0, 1.2 or 1.3 for the sx1302_hal.
Are you sure that the power supply is ok ?

chaudhariatul, 2022-12-01:
@mcoracin
Gateway hardware : It's an used RakV2 helium miner that I got from ebay. It was on the deny list of Helium. I'm repurposing this for something else.
Where should I look for the HAL details?
Power supply seems to be ok, I connected it to another setup of RaspberryPi with RAK2287 and it works. 

---
First run after power cycle (shutdown + wait 5mins + power on)
```
../util_chip_id/chip_id
CoreCell reset through GPIO17...
SX1261 reset through GPIO17...
CoreCell power enable through GPIO18...
CoreCell ADC reset through GPIO13...
Opening SPI communication interface
Note: chip version is 0x10 (v1.0)
INFO: using legacy timestamp
ARB: dual demodulation disabled for all SF
INFO: no temperature sensor found on port 0x39
INFO: no temperature sensor found on port 0x3B
INFO: no temperature sensor found on port 0x38
ERROR: no temperature sensor found.
ERROR: failed to start the gateway
```

Retrying after 10secs
```
../util_chip_id/chip_id
CoreCell reset through GPIO17...
SX1261 reset through GPIO17...
CoreCell power enable through GPIO18...
CoreCell ADC reset through GPIO13...
Opening SPI communication interface
Note: chip version is 0x10 (v1.0)
ERROR: Failed to set SX1250_0 in STANDBY_RC mode
ERROR: failed to setup radio 0
ERROR: failed to start the gateway
```

After 5 mins
```
../util_chip_id/chip_id
CoreCell reset through GPIO17...
SX1261 reset through GPIO17...
CoreCell power enable through GPIO18...
CoreCell ADC reset through GPIO13...
Opening SPI communication interface
Note: chip version is 0x05 (v0.5)
ERROR: Failed to set SX1250_0 in STANDBY_RC mode
ERROR: failed to setup radio 0
ERROR: failed to start the gateway
```

chaudhariatul, 2022-12-02:
Update:

Changing reset port to 25 and disabling temperature checks as done here helped: https://github.com/brocaar/sx1302_hal

I will need to create or find a patch for BasicStation when using RAK2287 : https://github.com/lorabasics/basicstation

mcoracin, 2022-12-02:
So, disabling the temperature sensor is to avoid the startup error I guess, right ?
Then the issue you were getting was due to the sx1302 not being  reset due to wrong pin, is that correct ?

chaudhariatul, 2022-12-02:
Correct.

RAK2287 board which comes from the Helium miner seems to be slightly different than the one sold separately. I cant tell what are these differences but looking from the top there is no GPS on the RakV2.

So setting the reset port to 25 works and shows the eui in output:

```
CoreCell reset through GPIO25...
SX1261 reset through GPIO25...
CoreCell power enable through GPIO18...
CoreCell ADC reset through GPIO13...
Opening SPI communication interface
Note: chip version is 0x10 (v1.0)
INFO: using legacy timestamp
ARB: dual demodulation disabled for all SF
INFO: no temperature sensor found on port 0x39
INFO: no temperature sensor found on port 0x3B
INFO: no temperature sensor found on port 0x38
ERROR: no temperature sensor found.

INFO: concentrator EUI: 0x0016c001f103b3b3

Closing SPI communication interface
CoreCell reset through GPIO25...
SX1261 reset through GPIO25...
CoreCell power enable through GPIO18...
CoreCell ADC reset through GPIO13...
```

I'm able to use the the board with thingstack:
```
cat docker-compose.yml
version: '3.7'
services:
  basicstation:
    image: xoseperez/basicstation:v2.5.0
    container_name: basicstation
    restart: unless-stopped
    privileged: true
    network_mode: host
    environment:
      MODEL: "SX1302"
      DEVICE: "/dev/spidev0.0"
      INTERFACE: "SPI"
      MODEL: "SX1302"
      DESIGN: "CORECELL"
      GW_RESET_GPIO: 25
      SPI_SPEED: "8000000"
      GATEWAY_EUI: "________"
      USE_CUPS: 0
      SERVER: "______"
      TLS_SNI: "false"
      TC_URI: "wss://_____:8887"
      TC_TRUST: "-----BEGIN CERTIFICATE-----MIIDvjCCAqagAwIBAgIUFOEN1....-----END CERTIFICATE-----"
      TC_KEY: "NNSXS...."
```

So the next thing to identify is a patch required to disable temperature. I have these references and need to find a right solution:

1. https://github.com/Lora-net/sx1302_hal/pull/63
2. https://github.com/lorabasics/basicstation/issues/157
3. https://github.com/lorabasics/basicstation/issues/125#issuecomment-934491017

Freshman0711, 2024-07-23:
我出现同样的问题，我已经得到了完美的解决，我的SX1302模块现在可以正常的工作，你可以尝试更改复位引脚和电源使能引脚，我用的是树莓派5，我用的是gpio23作为复位引脚，gpio22作为电源使能引脚，然后重启你的树莓派就可以解决，我建议切断电源防止模块没有得到断电复位。

PANJUN-T, 2025-04-07:
我也遇到了同样的问题，我在我的板子上找到的问题的原因：树莓派GPIO与实际寄存器索引不对应。需要修改reset_lgw.sh文件里的引脚序号
第一步，使用命令cat /sys/kernel/debug/gpio查看树莓派GPIO序号对应的寄存器索引号
hana@raspi:~ $ cat /sys/kernel/debug/gpio
gpiochip0: GPIOs 512-569, parent: platform/fe200000.gpio, pinctrl-bcm2711:
 gpio-512 (ID_SDA              )
 gpio-513 (ID_SCL              )
 gpio-514 (GPIO2               )
 gpio-515 (GPIO3               )
 gpio-516 (GPIO4               )
 gpio-517 (GPIO5               )
 gpio-518 (GPIO6               )
 gpio-519 (GPIO7               |spi0 CS1            ) out lo ACTIVE LOW
 gpio-520 (GPIO8               |spi0 CS0            ) out hi ACTIVE LOW
 gpio-521 (GPIO9               )
 gpio-522 (GPIO10              )
 gpio-523 (GPIO11              )
 gpio-524 (GPIO12              )
 gpio-525 (GPIO13              |sysfs               ) out hi
 gpio-526 (GPIO14              )
 gpio-527 (GPIO15              )
 gpio-528 (GPIO16              )
 gpio-529 (GPIO17              )
 gpio-530 (GPIO18              |sysfs               ) out hi
 gpio-531 (GPIO19              )
 gpio-532 (GPIO20              )
 gpio-533 (GPIO21              )
 gpio-534 (GPIO22              |sysfs               ) out hi
 gpio-535 (GPIO23              |sysfs               ) out lo
 gpio-536 (GPIO24              )
 gpio-537 (GPIO25              )
 gpio-538 (GPIO26              )
.....................

第二步修改reset_lgw.sh，根据硬件原理图我的Reset引脚是GPIO23，所以修改为535。你们根据实际情况修改
`#SX1302_RESET_PIN=23     # SX1302 reset
#SX1302_POWER_EN_PIN=18  # SX1302 power enable
#SX1261_RESET_PIN=22     # SX1261 reset (LBT / Spectral Scan)
#AD5338R_RESET_PIN=13    # AD5338R reset (full-duplex CN490 reference design)

SX1302_RESET_PIN=535     # SX1302 reset
SX1302_POWER_EN_PIN=530  # SX1302 power enable
SX1261_RESET_PIN=534     # SX1261 reset (LBT / Spectral Scan)
AD5338R_RESET_PIN=525    # AD5338R reset (full-duplex CN490 reference design)`