Lora-net/sx1302_hal #69: Hardware pins

kahlenberg, 2022-01-07:
Hi,
I have some questions regarding to Semtech SX1302CSSXXXGW1. I couldn't find any proper palce to ask my questions, sorry for inconviences. 

In the schematics, there are some pins to control the radio chips. If only SPI communication is used, where to connect those pins?
For example, there is a signal **POWEREN**. To which pin of Raspberry must it be connected when using SPI?
In the https://github.com/Lora-net/sx1302_hal/blob/master/tools/reset_lgw.sh#L16 there is a define **SX1302_POWER_EN_PIN**=18, but I am not sure if it is the same signal with **POWEREN**.

Where to connect those pins: **SX1302GPIO8**, **SX1261_DIO1**, **SX1261_BUSY**? 

**SX1261_NRESET** is clear in the script:
https://github.com/Lora-net/sx1302_hal/blob/master/tools/reset_lgw.sh#L17

**SX1261_NSS** is also clear:
https://github.com/Lora-net/sx1302_hal/blob/master/packet_forwarder/global_conf.json.sx1250.EU868#L14 

Thanks.

nilesh-dryad, 2022-01-07:
POWER_EN is connected to PIN 12 of raspberry pi header.

kahlenberg, 2022-01-07:
Thanks for answer.
That means, **POWER_EN** in schematics is **SX1302_POWER_EN_PIN** in script?
PIN12 is GPIO18 on the RPi Header. **SX1302_POWER_EN_PIN** actually means GPIO number not the PIN number. Some weird naming here. And it also activates the power for SKY chip, not SX1302.

Where can I find all and full information?
Thanks.

nilesh-dryad, 2022-01-17:
Sorry for late Answer. It is PIN number on raspberry pi header. Not the GPIO number,

closed 2022-09-30 by smtc-bot

techsd, 2023-04-17:
> Thanks for answer. That means, **POWER_EN** in schematics is **SX1302_POWER_EN_PIN** in script? PIN12 is GPIO18 on the RPi Header. **SX1302_POWER_EN_PIN** actually means GPIO number not the PIN [...]

i have problem with SPI and reset_lgw.sh