Lora-net/sx1302_hal #71: sx1302 and sx1250 low RSSI

vikasrohal, 2022-01-17:
Hi,
I'm testing a SX1302 + SX1250 based gateway with 5dBm o/p LoRa device 
20 dB Attenuator is connected between LoRa device and Gateway (This is conducted test)
I'm getting -3dBm signal on feeding point of SX1250 pin(RFI pin , measured by spectrum Analyzer) but packet forwader showing received RSSI is -33.
**Why that much deviation?**
I did the same with sx1301 + sx1257 based gateway,  its showing accepted results (-3dB received and showing near about -3 received RSSI)

vikasrohal, 2022-01-20:
Thats was a testing issue

closed 2022-01-20 by vikasrohal

mjlaurent6, 2023-03-09:
Hi can you share more about your experience ? I am having the issue on Sx1302 that receive very low rssi

saffetblt, 2023-08-24:
Hello,
I have a similar problem. I can't get more than -115 rssi with a gateway with SX1302. I have an external commercial gateway. But it can catch even -138 rssi values. Do you have any idea?

Thanks a lot.