Lora-net/sx1302_hal #80: Packet Forwarder configuration file for IN865

PrashKmrIn, 2022-08-01:
Hi, is there any configuration file for Indian LoRaWAN license free band 865-867MHz, or what all configuration do we need to set in existing configuration file in order to use at license free band in India?

mjlaurent6, 2023-02-02:
Hi any updates ?

henriBusch, 2024-01-23:
Hello, I would like my gateway to be reachable from end nodes from 433.05 to 434.79 MHz. The endnode a WL55jc2 can already do this by itself in the settings. Is it possible to take the EU868 and change all parameters to a EU433. 

Would this be possible by simply taking the global_conf.json.sx1250.EU868 and changing the individual parameters there. 

In the best case, maybe this already exists?

Many thanks in advance