Lora-net/sx1302_hal #89: DevEUI and AppKey inside packet-forwarder (payload structure)

sergorl, 2022-11-14:
Hello, guys. I work with [packet-forwarder](https://github.com/Lora-net/sx1302_hal) and wonder about structure of payload here:

```C
struct lgw_pkt_rx_s {
    uint32_t    freq_hz;        /*!> central frequency of the IF chain */
    uint8_t     if_chain;       /*!> by which IF chain was packet received */
    uint8_t     status;         /*!> status of the received packet */
    uint32_t    count_us;       /*!> internal concentrator counter for timestamping, 1 microsecond resolution */
    uint8_t     rf_chain;       /*!> through which RF chain the packet was received */
    uint8_t     modulation;     /*!> modulation used by the packet */
    uint8_t     bandwidth;      /*!> modulation bandwidth (LoRa only) */
    uint32_t    datarate;       /*!> RX datarate of the packet (SF for LoRa) */
    uint8_t     coderate;       /*!> error-correcting code of the packet (LoRa only) */
    float       rssi;           /*!> average packet RSSI in dB */
    float       snr;            /*!> average packet SNR, in dB (LoRa only) */
    float       snr_min;        /*!> minimum packet SNR, in dB (LoRa only) */
    float       snr_max;        /*!> maximum packet SNR, in dB (LoRa only) */
    uint16_t    crc;            /*!> CRC that was received in the payload */
    uint16_t    size;           /*!> payload size in bytes */
    uint8_t     payload[256];   /*!> buffer containing the payload */
};
```
My question: where is the `DevEUI` and `AppKey` inside payload of `struct lgw_pkt_rx_s `?

mcoracin, 2022-11-14:
Hello, the gateway doesn't have this information, and doesn't need it. It just forward the received packets to the LoRaWAN Network Server.

closed 2022-11-14 by mcoracin

sergorl, 2022-11-21:
> Hello, the gateway doesn't have this information, and doesn't need it. It just forward the received packets to the LoRaWAN Network Server.

I understand it. But where is DevEUI location inside payload?