Lora-net/sx1302_hal #93: Lora Phy packet receiving.

vAnArhist, 2022-12-29:
Hi all! Need help. 

- [ ] I ported this lib to other MCU chip (build from sources).
- [ ] I have a sensor, which sends **raw** LoRa packet every `1sec` on **915MHz**, **SF7** to SX1302/(2xSX1250) GW. 
- [ ] As I understand from API, there is no chance to receive data in interrupt handler (only from polling mode, as i see in [test_loragw_hal_rx.c](https://github.com/Lora-net/sx1302_hal/blob/master/libloragw/tst/test_loragw_hal_rx.c)), so I run this polling example on my platform.

Questions:
1. Don`t understand why max_rx_pkt = 16. 
``` C
/* set the buffer size to hold received packets */
    struct lgw_pkt_rx_s rxpkt[max_rx_pkt]; // max_rx_pkt = 16
```
2. With my setup I am able to receive something on this channel, but not every second, even not every minute. Any suggestions why?
3. It is not clear for me FIFO buffer size in SX1302.
4. Is it possible to receive/transmit RAW LoRa packets at all, or only LoRaWAN? If yes - how?
5. Is there any additional documentation for SX1302 than datasheet? Reason why I am asking - loragw lib is not clear without any HW info.

mcoracin, 2023-01-04:
Hello,
1. This size is arbitrary, it just needs to be big enough to hold the received packets depending on polling speed. If polling speed is faster than transmission speed on the end-device side, a size of 9 would be enough (for 9 channels).
2. Have you tried the same on a Raspberry Pi based gateway ? It could be a privite/public syncword issue ? If your sensor is sending packets with "private" syncword, the gateway configured for "public" syncword could still receive some of the packets...
3. The sx1302 can hold 4096 bytes in its FIFO. It can hold as many packets as the max number of bytes allows.
4. Yes it is possible, the gateway only cares about LoRa packets, the LoRaWAN part is only in the payload for the Network Server.
5. For a porting, you should only need to port the interfaces (SPI/USB/UART) parts of the code. You should not need to change anything else in the libloragw library.

May I ask on which MCU you are porting the lib ?

Best regards,
Michael

nilesh-dryad, 2023-01-04:
There is a way to receive Lora Packets on interrupts. It is not documented anywhere but you will get pulses on GPIO4 and GPIO6 when Lora Packet is received.
GPIO4 will get one pulse on each packet and GPIO6 will be toggled on each lora packet.

vAnArhist, 2023-01-05:
@mcoracin , @nilesh-dryad Really thank You for your clarifications! 
@mcoracin 
1. I ported it to ESP32S3 module. Indeed, changed only SPI parts of 1250/1302 communication and also aux timings functions.
SPI periph seems to work OK from tests (registers R/W, burst R/W).
2. I hope that on RPi-based gateway this lib will work without any issues, but I don`t have any chance to check, because we have custom HW.
3. It cannot be private/public syncword, SX1262-based sensor sends with same "public" syncword, as well as GW configured.
I will update You as soon as I will find the reason.

One more question: `sx1302_agc_load_firmware` and `sx1302_arb_load_firmware`. This 2 methods loads some FW I guess through SX1302 to 2 RF frontend(in my case SX1250). Question here - it loads to Flash or RAM memory (it should be flashed everytime or no)? 
The reason why I am asking is memory footprint. It consumes a lot of stack memory, so I have suggestion to flash that only once.

nilesh-dryad, 2023-01-05:
@vAnArhist We also ported to more than 2 MCU and we also faced this stack issue. You need to make those fw arrays as const so that it won't use stack memory.

vAnArhist, 2023-01-05:
@mcoracin @nilesh-dryad I`m really sorry, issue was in syncword preferences. I was completely convinced that sensors used "public" syncword.
Thank You for your patient and support! Stay safe.

closed 2023-01-05 by vAnArhist

mcoracin, 2023-01-05:
@vAnArhist the AGC and ARB firmwares are loaded into sx1302 internal memory, and it needs to be done at each startup.

Happy to know that your porting is now working. :)