Lora-net/sx1302_hal #94: Testing (test_loragw_hal_rx & test_loragw_hal_tx) error

amchangama, 2023-01-04:
Hi All
I am running the my experiment on Raspberry 4B and I could register the gate to the ttn, however when I tried to run the (test_loragw_hal_rx and  test_loragw_hal_tx) I got this error (ERROR:  failed to configure rxrf 0 ). 
Would you please  help in this. 

Kinds regards

mcoracin, 2023-01-04:
Could you please share the complete console output, and the command line parameters you used ?

amchangama, 2023-01-04:
> Could you please share the complete console output, and the command line parameters you used ?

[image]

vinayakpradhan, 2023-01-11:
Same issue here. When i run the files same output

mcoracin, 2023-01-12:
Yes, because you need to give some parameters to configure the hardware, and the modulation.
Using the -h param will give you the options available.

You mention TTN, but if you need to connect your gateway to the network server, you'll need to use the packet forwarder with the global_conf.json file corresponding to your region.

amchangama, 2023-01-12:
@mcoracin Thank your comment, would you mind giving  an example for t

> Yes, because you need to give some parameters to configure the hardware, and the modulation. Using the -h param will give you the options available.
>
> You mention TTN, but if you need to connect your [...]

@mcoracin  would you mind giving an example as that can pave the way for us to discover more. 
Thanks in advance

mcoracin, 2023-01-12:
If you want to connect a packet forwarder to TTN, you'll need to run (for EU868 region):

`./lora_pkt_fwd -c global_conf.json.sx1250.EU868.USB`

But before that update the global_conf.json.sx1250.EU868.USB file with your Gateway ID, server address and uplink/downlink ports as defined by TTN.

amchangama, 2023-01-13:
@mcoracin Thank you so much. It worked. 
However I am putting the comment for the spi version in case there is a someone who may need it :
`./test_loragw_hal_rx  -d /dev/spidev0.0  -k 0 -r 1250 -a 867.5 -b 868.5 -m 0`

closed 2023-01-16 by amchangama

techsd, 2023-04-18:
> @amchangama please find below examples for launching the HAL RX & TX test tools:
> 
> `./test_loragw_hal_rx -u -d /dev/ttyACM0 -k 0 -r 1250 -a 867.5 -b 868.5 -m 0`
> 
> => To be used for a USB corecell using sx1250 radios. It will initialize 8 channels as defined in the utility source code. 4 channels around the center frequency specified by -a param, and 4 other channels around the center frequency specified by -b param.
> 
> `./test_loragw_hal_tx -u -d /dev/ttyACM0 -k 0 -r 1250 -f 868.5 -m LORA -s 9 -b 125 -l 8 -n 10 -z 23 -i`
> 
> => For sending 10 LoRa packets of size 23 bytes with SF9 BW125 KHz.

when i start  lora_pkt_fwd -c global_conf.json.sx1250.EU868.USB
i have 
*** Packet Forwarder ***
Version: 2.1.0
*** SX1302 HAL library version info ***
Version: 2.1.0;
***
INFO: Little endian host
ERROR: [main] failed to find any configuration file named global_conf.json.sx1250.EU868.USB

techsd, 2023-04-18:
> @mcoracin Thank you so much. It worked. However I am putting the command for the spi version in case there is a someone who may need it : `./test_loragw_hal_rx -d /dev/spidev0.0 -k 0 -r 1250 -a 867.5 -b 868.5 -m 0`

 test_loragw_hal_rx  -d /dev/spidev0.0  -k 0 -r 1250 -a 867.5 -b 868.5 -m 0
===== sx1302 HAL RX test =====
INFO: rxpkt buffer size is set to 16
INFO: Select channel mode 0
CoreCell reset through GPIO23...
SX1261 reset through GPIO23...
CoreCell power enable through GPIO18...
CoreCell ADC reset through GPIO13...
Opening SPI communication interface
Note: chip version is 0x00 (v0.0)
ERROR: Failed to set SX1250_0 in STANDBY_RC mode
ERROR: failed to setup radio 0
ERROR: failed to start the gateway