Lora-net/sx1302_hal #101: MultiSF channels ignores implicit header mode for RX

nsforth, 2023-03-15:
Hi!

I recently tried to use SF6 with station based on SX1302. There is a problem with SX1276 based devices that require implicit header mode for SF6.
SF6 and implicit header only work for LoraStd modem, 8x8 multiSF channels simply ignore implicit header settings in ifcfg struct.
Any packets sent by sx1276 with implicit header received by LoraStd, multiSF channels never seen SF6 packets.

Is it really possible to use multiSF 8x8 channels with implicit header mode?

mcoracin, 2023-03-15:
Hello,
Implicit header is not supported by multiSF channels, only the LoRaStd (called lora_service in sx1302 HAL) supports implicit header.

It is indicated in the HAL here:
https://github.com/Lora-net/sx1302_hal/blob/4b42025d1751e04632c0b04160e0d29dbbb222a5/libloragw/inc/loragw_hal.h#L203

Best regards,
Michael

nsforth, 2023-03-15:
Many thanks for fast answer!

Ok, but for what purpose exist registers in loragw_reg.h like SX1302_REG_RX_TOP_TXRX_CFG2_IMPLICIT_HEADER and similar registers there:  https://github.com/Lora-net/sx1302_hal/blob/4b42025d1751e04632c0b04160e0d29dbbb222a5/libloragw/inc/loragw_reg.h#L602
Registers names looks like something useful for RX implicit header configuration. :/

mcoracin, 2023-03-15:
Indeed, you're right.
It is possible, but it has never been tested. Also not sure about compatibility with sx1276 SF6.

But you can give it a try, for this you'll need to set the following registers:
- SX1302_REG_RX_TOP_TXRX_CFG2_IMPLICIT_HEADER
- SX1302_REG_RX_TOP_TXRX_CFG1_CODING_RATE
- SX1302_REG_RX_TOP_TXRX_CFG2_CRC_EN
- SX1302_REG_RX_TOP_TXRX_CFG3_PAYLOAD_LENGTH

You may need to disable the fine sync for SF6 also by setting the following register to false:
- RX_TOP_TXRX_CFG9_FINE_SYNCH_EN_SF6

So you realize that the packet length is fixed for all channels in this case.

Again, it has never been tested, and it is not officially supported, so we may not be able to provide much more support than this.

sx1261 radio provides a better SF5/SF6 support.

nsforth, 2023-03-15:
It's okay for me. 
Should i disable fine sync before packet send or it must be also disable on multisf modems configuration somewhere in lgw_start stage?

nsforth, 2023-09-01:
Even with your last recommendations it doesn't work. Anyway, thanks a lot!