Lora-net/sx1302_hal #103: Can't get gateway EUI

YegorRad, 2023-04-28:
Hello everyone

I'm following this instructions in order to get started with LoraWan: https://www.waveshare.com/wiki/SX1302_LoRaWAN_Gateway_HAT#Documents

The issue is that when I try to get the gateway EUI using the command "./chip_id" it appears this:

**admin@raspberrypi:~/Documentos/sx1302_hal/util_chip_id $ ./chip_id 
CoreCell reset through GPIO23...
SX1261 reset through GPIO23...
CoreCell power enable through GPIO18...
CoreCell ADC reset through GPIO13...
Opening SPI communication interface
ERROR: failed to start the gateway**

Any solutions?
Thanks in advance

tfal21, 2023-04-28:
i have the same issue but i have the 1303 chipset, so i thought that was the problem, but to them its the same

YegorRad, 2023-05-02:
> i have the same issue but i have the 1303 chipset, so i thought that was the problem, but to them its the same. my error looks like this corecell reset through gpio23 sx1261 reset through gpio23 corecell power enable through gpio18 corecell adc reset through gpio13 opening spi communications interface note: chip version is 0x00 (v0.0) error failed to set sx1250_0 in standby_rc mode error failed to setup radio 0 error failed to start gateway
> 
> i think the chip_id program is not current enough to identify sx1303 chips

YegorRad, 2023-05-02:
> i have the same issue but i have the 1303 chipset, so i thought that was the problem, but to them its the same. my error looks like this corecell reset through gpio23 sx1261 reset through gpio23 [...]

After reading a lot about this issue I found that, in most cases, the problem is related to reset pins configuration. I've opened SPI communication, setted SX1302_RESET_PIN=17, opened the Serial Port (which is disabled by default)...

Still having the problem...

I'm trying to configurate the Gateway but I'm having this error now:

admin@raspberrypi:~/Documentos/sx1302_hal/packet_forwarder $ sudo ./lora_pkt_fwd -c global_conf.json.sx1250.EU868
***** Packet Forwarder ***
Version: 2.1.0
*** SX1302 HAL library version info ***
Version: 2.1.0;
***
INFO: Little endian host
INFO: found configuration file global_conf.json.sx1250.EU868, parsing it
INFO: global_conf.json.sx1250.EU868 does contain a JSON object named SX130x_conf, parsing SX1302 parameters
INFO: com_type SPI, com_path /dev/spidev0.0, lorawan_public 1, clksrc 0, full_duplex 0
INFO: antenna_gain 0 dBi
INFO: Configuring legacy timestamp
INFO: Configuring Tx Gain LUT for rf_chain 0 with 16 indexes for sx1250
INFO: radio 0 enabled (type SX1250), center frequency 867500000, RSSI offset -215.399994, tx enabled 1, single input mode 0
INFO: radio 1 enabled (type SX1250), center frequency 868500000, RSSI offset -215.399994, tx enabled 0, single input mode 0
INFO: Lora multi-SF channel 0>  radio 1, IF -400000 Hz, 125 kHz bw, SF 5 to 12
INFO: Lora multi-SF channel 1>  radio 1, IF -200000 Hz, 125 kHz bw, SF 5 to 12
INFO: Lora multi-SF channel 2>  radio 1, IF 0 Hz, 125 kHz bw, SF 5 to 12
INFO: Lora multi-SF channel 3>  radio 0, IF -400000 Hz, 125 kHz bw, SF 5 to 12
INFO: Lora multi-SF channel 4>  radio 0, IF -200000 Hz, 125 kHz bw, SF 5 to 12
INFO: Lora multi-SF channel 5>  radio 0, IF 0 Hz, 125 kHz bw, SF 5 to 12
INFO: Lora multi-SF channel 6>  radio 0, IF 200000 Hz, 125 kHz bw, SF 5 to 12
INFO: Lora multi-SF channel 7>  radio 0, IF 400000 Hz, 125 kHz bw, SF 5 to 12
INFO: Lora std channel> radio 1, IF -200000 Hz, 250000 Hz bw, SF 7, Explicit header
INFO: FSK channel> radio 1, IF 300000 Hz, 125000 Hz bw, 50000 bps datarate
INFO: global_conf.json.sx1250.EU868 does contain a JSON object named gateway_conf, parsing gateway parameters
INFO: gateway MAC address is configured to AA555A0000000000
INFO: server hostname or IP address is configured to "localhost"
INFO: upstream port is configured to "1730"
INFO: downstream port is configured to "1730"
INFO: downstream keep-alive interval is configured to 10 seconds
INFO: statistics display interval is configured to 30 seconds
INFO: upstream PUSH_DATA time-out is configured to 100 ms
INFO: packets received with a valid CRC will be forwarded
INFO: packets received with a CRC error will NOT be forwarded
INFO: packets received with no CRC will NOT be forwarded
INFO: GPS serial port path is configured to "/dev/ttyS0"
INFO: Reference latitude is configured to 0.000000 deg
INFO: Reference longitude is configured to 0.000000 deg
INFO: Reference altitude is configured to 0 meters
INFO: Beaconing period is configured to 0 seconds
INFO: Beaconing signal will be emitted at 869525000 Hz
INFO: Beaconing datarate is set to SF9
INFO: Beaconing modulation bandwidth is set to 125000Hz
INFO: Beaconing TX power is set to 14dBm
INFO: Beaconing information descriptor is set to 0
INFO: global_conf.json.sx1250.EU868 does contain a JSON object named debug_conf, parsing debug parameters
INFO: got 2 debug reference payload
INFO: reference payload ID 0 is 0xCAFE1234
INFO: reference payload ID 1 is 0xCAFE2345
INFO: setting debug log file name to loragw_hal.log
INFO: [main] TTY port /dev/ttyS0 open for GPS synchronization
CoreCell reset through GPIO17...
SX1261 reset through GPIO17...
CoreCell power enable through GPIO18...
CoreCell ADC reset through GPIO13...
Opening SPI communication interface
Note: chip version is 0x00 (v0.0)
ERROR: Failed to set SX1250_0 in STANDBY_RC mode
ERROR: failed to setup radio 0
ERROR: [main] failed to start the concentrator**

HCovas, 2023-05-12:
Hello everyone,

I'm also using the same Waveshare Hat and faced similar errors. Besides the reset pins, you need to have the SPI and I2C interfaces also enabled. 

However, with everything enabled and all the pins configured, I was still having problems starting the gateway. I've contacted Waveshare which insisted that my SX1303 mini-pcie module was not "making a good contact" in the connector. I couldn't understand, the module was correctly inserted in the connector.
 After a lot of troubleshoot and emails exchanged with Waveshare, I've found out that leaving the module hanging/tilted without clipping it, it works correctly. See the image attached. 

it seems strange, but if I put it in the correct position, I get the same previous errors. It's for sure a problem with waveshare hat mini-pcie connector. 

Give a try, and see if it helps.
Cheers.
 

[image]

tfal21, 2023-05-12:
Amazing, that worked for me as well.  Out of curiosity, In the end, which gateway software did you end up using?  Waveshare, lorabasic station, or other?
Thanks

HCovas, 2023-05-15:
Great. I'm glad it helped. 

I think most of the problems people are having with this HAT may be solved with this "trick". Although this is a Waveshare problem and they should be held accountable for that.

I'm currently using the Lora-net/sx1302_hal as recommended by waveshare in their wiki. It's integrated with TTN and I'm receiving the gateway status correctly in the server. However I'm not getting the data from the nodes (SX126X). Still working on it.  

If I'm not able to make it work, maybe I'll try the Lora basics Station (https://github.com/lorabasics/basicstation).

Cheers.

mcoracin, 2023-05-15:
Hi @HCovas ,

The Lora-net/sx1302_hal project should be more than enough to allow you to connect your nodes to TTN network.
Which kind of issue are you facing to connect your end-nodes ? Which LoRaWAN stack are you running on it ?

Best regards,
Michael

HCovas, 2023-05-15:
Thank you so much @mcoracin. It worked. 
just changed the gateway configuration to `"lorawan_public": false,` and started working perfectly fine. 

Thanks again. 
Cheers.

Kitarp00, 2023-06-16:
@HCovas Can you help me with the setup of the sx1303. I have exact same setup. but I am able to get it working. even after keeping the module tilted

HCovas, 2023-06-16:
Hey @Kitarp00, what kind of error are you having? Did you follow the setup from the Waveshare page? 

Cheers

Kitarp00, 2023-06-16:
Hi @HCovas, I followed the wiki of the waveshare and landed to this error.
When I run 
`./chip_id`
```
CoreCell reset through GPIO17...
SX1261 reset through GPIO17...
CoreCell power enable through GPIO18...
CoreCell ADC reset through GPIO13...
Opening SPI communication interface
Note: chip version is 0x12 (v1.2)
ERROR: Failed to set SX1250_0 in STANDBY_RC mode
ERROR: failed to setup radio 0
ERROR: failed to start the gateway
```

I also changed the SX1302_RESET_PIN=17     # SX1302 reset
What should I do next?

HCovas, 2023-06-19:
Hello @Kitarp00 , 

Did you setup the reset pins on the reset_lgw.sh, right? 
I really suspect it is an hardware problem, although you are getting the chip version. Which is strange. 
Try to better insert the module into the connector, force it a little but always leave it inclined. You have to try it out a few times, and test slightly different module positions.

Also, make sure you have all the SPI and I2C correctly enabled. 

My reset pins on reset_lgw.sh file are different because I'm not using a RPi4, I'm using a Rock Pi4 C+. The hat connector has the same pinout of the raspberry. But the pin assignment is different.

Don't know if it helps, but I leave the pins that I'm using here below. 

SX1302_RESET_PIN=154 # SX1302 reset (Pin 16)
SX1302_POWER_EN_PIN=131 #SX1302 power enable (Pin 12)
SX1261_RESET_PIN=149 #SX1261 reset (LBT/Spectral Scan) (Pin 15)
AD5338R_RESET_PIN=13  # AD5338R reset 

Don't know if helps. 
Let me know. 

Cheers.