Lora-net/sx1302_hal #104: About tx failed

JenkinTu, 2023-04-29:
SX1302 has certain propability less than 10% failed to tx packets. As the following log, the packet should be transmitted when 9:18:11, but it had been replaced by the packet should be transmitted when 9:18:22. This packet had been written into SX1302 registers ahead of 38ms before transmission. Why hadn't this packet been transmitted?

---------------------------------------------------------------------------------------------------------------------------------------------
4-29 9:18:10 [/usrdata/log/pktfwd] INF thread_down:3219 JSON down: {"txpk":{"imme":false,"rfch":0,"powe":17,"ant":0,"brd":0,"tmst":3624665456,"freq":475.5,"modu":"LORA","datr":"SF7BW125","codr":"4/5","ipol":true,"size":25,"data":"YDQAACSA+U8CytoOJOBBIHsN/Eyouzmo8w=="}}
4-29 9:18:11 [/usrdata/log/pktfwd] INF thread_jit:3764 lgw_send done on rf_chain 0: count_us=3624665456, contractor_time= 3624627141
...

4-29 9:18:22 [/usrdata/log/pktfwd] INF thread_down:3219 JSON down: {"txpk":{"imme":false,"rfch":0,"powe":17,"ant":0,"brd":0,"tmst":3636165503,"freq":476.7,"modu":"LORA","datr":"SF7BW125","codr":"4/5","ipol":true,"size":25,"data":"YCAAACSAGFUC9QccIlsmFO2gTAba26fXSQ=="}}
4-29 9:18:22 [/usrdata/log/pktfwd] INF thread_jit:3737 WARNING: a downlink was already scheduled on rf_chain 0, overwritting it...
4-29 9:18:22 [/usrdata/log/pktfwd] INF print_tx_status:3674 INFO: [jit] lgw_status returned TX_SCHEDULED
4-29 9:18:22 [/usrdata/log/pktfwd] INF thread_jit:3764 lgw_send done on rf_chain 0: count_us=3636165503, contractor_time= 3636126593
......

JenkinTu, 2023-05-03:
We are using ChirpStack. The TX packet had already been sent from ChirpStack to gateway, and properly been enqueued into jit queue. 
Ahead of 38ms the TX packet had been written into SX1302 tx buffer before transmission.  Should we use TX_JIT_DELAY some bigger value than 40ms?

https://github.com/Lora-net/sx1302_hal/blob/4b42025d1751e04632c0b04160e0d29dbbb222a5/packet_forwarder/src/jitqueue.c#L422-L428

mcoracin, 2023-05-03:
Yes, you can try to increase the TX_JIT_DELAY to better match with your system latencies.