Lora-net/sx1302_hal #107: Failure regarding setting up sx_1250 interfacing

satwik007sundaram, 2023-05-27:
I am trying to setup the sx_1303 LoraWAN gateway as given in the official documentation. I have created my TTN account and trying to execute chip_id file but am getting following error:

CoreCell reset through GPI023... SX1261 reset through GPIO23...

CoreCell power enable through GPIO18...

CoreCell ADC reset through GPIO13... Opening SPI communication interface

Note: chip version is 0x00 (v0.0) ERROR: Failed to set SX1250_0 in STANDBY_RC mode

ERROR: failed to setup radio 0 ERROR: failed to start the gateway

Kitarp00, 2023-06-16:
Facingg same issue

Shlabder, 2024-05-24:
Anyone found a fix for this issue ?

mcoracin, 2024-05-27:
Hello,
Could you provide more details about the hardware you are using? Which sx1302 board, which host (Rpi I guess)...
Thanks