Lora-net/sx1302_hal #113: chip version is 0x00 and all the registers are 0x00 with Raspberry 3B+

workholic365, 2023-08-17:
Hi ,
    Recently I got a sx1302 module with raspberry transfer board , I try to execute with compiling the open source sx1302_hal codes .I tried to test with test_loragw_reg , the result as follow  :

> CoreCell reset through GPIO23...
> SX1261 reset through GPIO23...
> CoreCell power enable through GPIO18...
> CoreCell ADC reset through GPIO13...
> Opening SPI communication interface
> open spidev0.0 OK [com_stat = 0]   #this is my test to check whether spidev0.0 could work
> Note: chip version is 0x00 (v0.0)

   I checked the reset pin ,it worked . the spi device could also effect .
   so does anyone met the same issue ? and  may give me some suggestions about it .appreciate your help .

workholic365, 2023-08-18:
Hi @mcoracin ,
    Thanks for your reply .
    yes ,I executed the reset_lgw.sh script which  copied from the tool path according to the instruction .  
    Do you have some suggestions for my current issue ? 
    Thanks !

mcoracin, 2023-08-18:
Which sx1302 board are you using exactly ?

workholic365, 2023-08-18:
it can be found in sx1302 module : should be made by greenpalm

workholic365, 2023-08-18:
Hi @mcoracin 
    I have tried ,the result as follow :

   t_loragw_com -d /dev/spidev0.0
CoreCell reset through GPIO23...
SX1261 reset through GPIO22...
CoreCell power enable through GPIO18...
CoreCell ADC reset through GPIO13...
Beginning of test for loragw_com.c
Opening SPI communication interface
spi tx buff is :
 0x 0  0x56  0x 6  0x 0  0x 0
spi rx buff is :
 0x 0  0x 0  0x 0  0x 0  0x 0
SX1302 version: 0x00
spi tx buff is :
 0x 0  0x57  0x80  0x 0  0x 0
spi rx buff is :
 0x 0  0x 0  0x 0  0x 0  0x 0
Cycle 0> error during the buffer comparison
Written values:
 63  F7  13  51  1E  E7  54  78  B5  43  F7  60  11  C2  3E  83
 C5  46  98  92  7C  CE  34  4D  F5  98  D9  38  4B  D5  92  AF
 CC  A5  00  EB  8C  54  63  41  97  5B  A2  A8  1D  E0  2B  E2
 26  C4  74  A2  92  A9  EF  88  41  C8  C0  8D  9D  52  3C  6A
 F8  3C  55  84  90  B8  C6  27  13  68  CF  30  48  FA  13  6E
 BE  87  10  51  30  FF  D9  72  C7  99
Read values:
 00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00
 00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00
 00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00
 00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00
 00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00
 00  00  00  00  00  00  00  00  00  00
Closing SPI communication interface
End of test for loragw_com.c

   it seemed fail to read . could you find something wrong from above result ?
    appreciate your help !Thanks !

workholic365, 2023-08-18:
I checked the reset by echo to make it set or reset ,it  worked .

mcoracin, 2023-08-21:
I can't say much from the logs, either your SPI connection is wrong, or the GPIO pin values are not set properly in reset_lgw.sh script for RESET and POWER_EN pins.

You are sure that this sx1302 board is an SPI version, and not a USB version, right ?

workholic365, 2023-08-25:
Hi @mcoracin,
    sorry to reply you late , I am doing some tests all these days .
   I think the issue may be caused by the spi pin , the real spi pin not match . another sw board could work ,have test with the lorawan node and open lorawan server , it seemed normal currently .
    Many thanks .

jroeber, 2023-11-25:
I'm having a similar issue. I used a WM1302-USB-915-M with no issues, but I had to downgrade to the slightly older WM1302-USB-915 (notice no *-M*), which does not work with the packet forwarder. Are the GPIO pins different between the two models?