Lora-net/sx1302_hal #122: Issue with Australian LoRa Setup - Request for Australia Configuration File in sx1302_hal Library

ABHITHLALC, 2024-04-29:
Hello everyone,

I'm encountering an issue with my LoRa setup in the Australia region. Here's my setup:

Device: ESP32 with SX1262 Waveshare module
Gateway: Raspberry Pi 4 with SX1303 915MHz, using the sx1302_hal library
Configuration File: Using the template from sx1302_hal library [global_conf.json.sx1250.US915](https://github.com/Lora-net/sx1302_hal/blob/master/packet_forwarder/global_conf.json.sx1250.US915)
TTN Cluster: Registered in Australia Network Cluster
Frequency Band: AU_915_928_FSB_2
Even though the gateway connects and runs successfully, my device can't join the network, and I'm not receiving any messages in TTN. Since I couldn't find a specific configuration file for Australia, I used the US 915 config file, thinking they would be similar enough to avoid issues. But now I'm beginning to suspect this might be causing the problem.

Here's what I've tried to resolve the issue:

Europe 868 MHz Configuration: I switched to a Waveshare SX1303 868MHz module, registered in the Europe Network Cluster, and adjusted the frequency band and gateway configuration accordingly. This setup worked, with LoRa frames being received by TTN.
US 915 MHz Configuration: I reverted to the original setup with SX1303 915MHz and registered in the US Network Cluster. Using the same global configuration file (global_conf.json.sx1250.US915), this setup also worked, with messages being received by TTN.
Given this information, I'm looking for advice on what might be causing the issue with my Australian setup. Is there something specific about the AU_915_928 frequency band that I might be missing? Could the gateway configuration or device settings be incorrect, or are there other environmental factors at play?

Additionally, I'd like to request a global configuration file specifically for Australia in the sx1302_hal library. This would greatly help developers like me who are setting up LoRa in this region and might be experiencing similar issues.

Any assistance or guidance would be greatly appreciated. I'm ready to provide additional information or context if needed.

Thank you in advance!

mcoracin, 2024-05-02:
Hello,
So on the end-device side, you have configured the LoRaWAN stack to operate using the AU_915_928_FSB_2 band, right ?
Have you updated the gateway json configuration file to operate on the same band configuration ?
The frequency plan is described here:
https://github.com/TheThingsNetwork/lorawan-frequency-plans/blob/master/US_902_928_FSB_2.yml
Best regards,
Michael

ABHITHLALC, 2024-05-03:
@mcoracin 
Yes, on the end-device side, I've configured the LoRaWAN stack to operate with the AU_915_928_FSB_2 band.
Does this work on both AU and US?

I'm curious if the [US_902_928_FSB_2](https://github.com/TheThingsNetwork/lorawan-frequency-plans/blob/master/US_902_928_FSB_2.yml) configuration could be used for both regions. While I understand that the Australian frequency range falls within the US spectrum, I want to ensure there's no misalignment or compliance issue.

Given these frequency plans, can the same configuration file work for both US and AU, or does Australia require a specific configuration to avoid conflicts?
[image]
[image]

ABHITHLALC, 2024-05-03:
Thank you for your response, I will check this.

bbelson2, 2024-06-07:
The following worked for me. Starting from `sx1302_hal/packet_forwarder/global_conf.json.sx1250.US915`, I created a TTN gateway, downloaded the TTN `global_conf.json` and made the following changes to `global_conf.json.sx1250.US915`:

```
"SX130x_conf" > "radio_0" > "freq" = 917200000
"SX130x_conf" > "radio_0" > "rssi_offset" = -166.0
"SX130x_conf" > "radio_0" > "tx_freq_min" = 915000000
"SX130x_conf" > "radio_0" > "tx_freq_max" = 928000000
"SX130x_conf" > "radio_1" > "freq" = 917900000
"SX130x_conf" > "radio_1" > "rssi_offset" = -166.0
"gateway_conf" > "gateway_ID" = "<gateway ID copied from the global_conf.json downloaded from TTN>"
"gateway_conf" > "server_address" = "<TTN server address copied from global_conf.json>"
"gateway_conf" > "serv_port_up" = 1700
"gateway_conf" > "serv_port_down" = 1700
"gateway_conf" > "servers" = <Entire section copied from global_conf.json. Remember to add a comma to the previous line.>
```
It's obviously a good idea to rename the resulting file, as well.

Hopefully this will help someone. 😄