Lora-net/sx1302_hal #124: SF5 and SF6 supported only for private 0x12 syncword

reissjason, 2024-06-17:
Would there be any issue changing SF5 and SF6 to use the public 0x34 syncword?

mcoracin, 2024-06-17:
Yes, the 0x34 syncword cannot be coded for SF5, and for SF6 the sensitivity is degraded. That's why we kept 0x12 for public for SF5/SF6 in the hal.

rsoss, 2024-06-17:
Hi @mcoracin does it mean that we must use the 0x12 (private syncword) for SF5/6? How would this interact with public syncword  (0x34) for SF7...12 on the same gateway?

mcoracin, 2024-06-17:
Yes, you must use 0x12 for SF5/SF6.
The syncword configuration is per SF so it is not an issue to have SF5-6 with 0x12, and SF7-12 with 0x34.
The end-device has to use the proper syncword depending on the SF used for transmission.