Lora-net/sx1302_hal #125: How to integrate with aws gateway sample configuration file

Meimurugan11, 2024-06-18:
Hi all,

I want to integrate the sx1302 lorawan packet forwarder with aws server it requires aws key, certifates, trust certificates.
I am trying to configure with sample configuration files am not t=sure this format is right.
Please help me to resolve this issue. 

sample configuration:

"gateway_conf": {
        "gateway_ID": "7cdbf83c-5a02-48fb-92cf-259d3c76c1ef",
        /* change with default server address/ports */
        "server_address": "A2QYF4M3Q8W7FT.cups.lorawan.ap-northeast-1.amazonaws.com",
        "serv_port_up": 443,
        "serv_port_down": 443,
        /* adjust the following parameters for your network */
        "keepalive_interval": 10,
        "stat_interval": 30,
        "push_timeout_ms": 100,
        /* forward only valid packets */
        "forward_crc_valid": true,
        "forward_crc_error": false,
        "forward_crc_disabled": false,
        /* GPS configuration */
        "gps_tty_path": "/dev/ttyS0",
        /* GPS reference coordinates */
        "ref_latitude": 0.0,
        "ref_longitude": 0.0,
        "ref_altitude": 0,
        /* Beaconing parameters */
        "beacon_period": 0,
        "beacon_freq_hz": 869525000,
        "beacon_datarate": 9,
        "beacon_bw_hz": 125000,
        "beacon_power": 14,
        "beacon_infodesc": 0,
        "servers": [
   "servers": [
      {
        "gateway_ID": "7cdbf83c-5a02-48fb-92cf-259d3c76c1ef",
        "server_address": "A2QYF4M3Q8W7FT.cups.lorawan.ap-northeast-1.amazonaws.com",
        "serv_port_up": 443,
        "serv_port_down": 443,
        "serv_enabled": true,
         "uri": "A2QYF4M3Q8W7FT.cups.lorawan.ap-northeast-1.amazonaws.com",
        "trust":"cups.trust",
        "key": "7cdbf83c-5a02-48fb-92cf-259d3c76c1ef.private.key",
        "cert": "7cdbf83c-5a02-48fb-92cf-259d3c76c1ef.cert.pem"
      }
    ]
    },

Thanks

mcoracin, 2024-06-18:
Hello,
As far as I know, the UDP packet forwarder cannot be connected to AWS LNS, unfortunately. Only lorabasicstation is supported with AWS.
If you want to use the UDP packet forwarder, you can use The Things Network, or The Things Stack, or Chirpstack and many other LoRa Network Servers (LorIoT, ...).
Almost all LNS support the packet forwarder, but not AWS. ;)

Meimurugan11, 2024-06-19:
Hi @mcoracin,

Thanks for your information.
i have lora basic station code as well but am not sure about the configuration in station.conf file.
am not sure how to figure out the issue in that & how to integrate with aws.
Any help would be greatly appreicated.

Thanks