Lora-net/sx1302_hal #129: How to receive downlinks from other gateways

warnbx, 2024-08-23:
I'm only getting uplinks from my end devices when I start the packet forwarder on my sx1302. But I also want to see the downlink replies from the infrastructure (separate gateway). Is there an option or configuration where I can enable capturing downlink packets as well?

**What is the goal:**
I want to capture all LoRa packets in my environment for debug purposes (all uplinks from end devices and the downlinks from the gateways). So I don't want to use my setup as a standalone gateway. It should just passively listen to the environment and print out all packets it concentrates over the air (like a sniffer).

**I use the following hardware:**
- Raspberry Pi 4B with debian 10 buster installed
- Dragino PG1302 shield with sx1302 chip

mcoracin, 2024-08-23:
Hello,

In LoRaWAN, uplinks and downlinks do not use the same polarity. Uplinks use non-inverted polarity, and downlinks use inverted polarity.
A LoRa demodulator will only receive packets with the configured polarity, so it is not possible to receive both uplinks and downlinks with the same demodulator.
The only thing that could be possible is to use the "lora_std/lora_service" channel, configured for inverted polarity to receive downlinks, but it would be on a single channel, with static datarate configuration.

Basically, it is not possible to capture the complete LoRaWAN uplink/traffic with a single gateway.

Best regards,
Michael

warnbx, 2024-08-23:
Thanks for your quick response! That helped a lot.

Is there a configuration for the polarity so that I can switch between capturing uplink and downlink packets? I am thinking of using two devices/boards, one capturing the uplink frames and the other the downlink frames. I need this to work on all channels, the single channel solution does not work for me.

mcoracin, 2024-08-23:
You can try the following:
Update the _sx1302_lora_modem_configure()_ function by adding the following lines before the return at the end.
````c
/* Chirp invert */
err |= lgw_reg_w(SX1302_REG_RX_TOP_RX_CFG0_CHIRP_INVERT, 0);
````
This should configure the 8 multi-SF channels to receive inverted polarity packets. (note that the register value meaning is reversed).

If you also want to configure the lora_std/lora_service channel (single SF), for example to receive the RX2 downlinks, you can update the _sx1302_lora_service_modem_configure()_ function by adding the following lines:

````c
/* Chirp invert */
err |= lgw_reg_w(SX1302_REG_RX_TOP_LORA_SERVICE_FSK_RX_CFG0_CHIRP_INVERT, 0);
````