Lora-net/sx1302_hal #135: Questions about loragw_sx1302_timestamp.c

ECherr, 2025-01-13:
Hi, my gateway use basicstation mode with class b enable. But recently I get some questions.

I gaint pps_xticks with 0 return sometime, and old value(same as last) sometime.

I chek the code about this, find that end with write/read to use fd.

So, I wander to know, how the sx1302 use the counters to save pps signal?

In `timestamp_counter_get` function which locate in loragw_sx1302_timestamp.c, read two counters value, one for freerun counter, another for pps counter, so , in sx1302, maintain two counters?

If call `timestamp_counter_get` function, sx1302 will hang up until pps is up edge, then return?

Could u tell me how they work?

Best Regards.

ECherr, 2025-01-13:
this return 0

[image]

and this return same value as last
[image]

mcoracin, 2025-01-13:
Hello,
The sx1302 has 1 internal counter from which we can get 2 values:
- the current free running value
- the value the counter was at last PPS
The sx1302_hal allows to read both values. But if the PPS is lost due to bad GPS signal quality, the sx1302 counter value latched on PPS will keep the same value, until a PPS comes in.
Best regards,
Michael

ECherr, 2025-01-13:
Thanks for your replay.

When we call some function, like `timestapm_counter_get`, will it make an interrupt on sx1302? And I foud that, sx1302 clock source from sx1250.

I test once again just now. In which condition, sx1302 will return `0`?  Does it mean that the PPS had lost before sx1302 power up?
If restart action on sx1302(`lgw_stop` and `lgw_start`), will it clean the internal counter values?

mcoracin, 2025-01-13:
No, there is no interrupt. When timestamp_counter_get() is called it just read registers from sx1302 where the current value of the internal counter is stored, and the value of this same counter at last PPS.

The sx1250 radio will give the 32M clock to the sx1302. If the sx1250 is not powered and started, the sx1302 free run counter will be 0. If the free run counter is not 0, but the PPS counter is 0, it means that there is no PPS signal sent to the sx1302.

lgw_stop/lgw_start will not clean the internal counter, the sx1302 needs to be reset.

Which hardware are you using ? Is it a SPI or USB sx1302 board ?

ECherr, 2025-01-13:
Ooo, many thanks.

I use RAK7289CV2, I think it is USB sx1302 board.

I met that pps_xticks is zero or old value, so I try to understand how the internal counter works in basicstation mode.

I have another question, if startup with good gps signal, work for a time, and lost gps signal for a long time, such as late about 30mins, then gain gps signal again, will sx1302 recovery the gps drift?