Lora-net/sx1302_hal #137: How to install on orange pi

hiteshtechshslok, 2025-01-28:
Hello,

I am trying to install the sx1302_hal on Orange pi and getting this error.

`radxa@radxa-cm3-io:~/sx1302_hal$ make clean all
make clean -e -C libtools
make[1]: Entering directory '/home/radxa/sx1302_hal/libtools'
rm -f libtinymt32.a
rm -f libparson.a
rm -f libbase64.a
rm -f obj/*.o
make[1]: Leaving directory '/home/radxa/sx1302_hal/libtools'
make clean -e -C libloragw
make[1]: Entering directory '/home/radxa/sx1302_hal/libloragw'
rm -f libloragw.a
rm -f test_loragw_*
rm -f obj/*.o
rm -f inc/config.h
make[1]: Leaving directory '/home/radxa/sx1302_hal/libloragw'
make clean -e -C packet_forwarder
make[1]: Entering directory '/home/radxa/sx1302_hal/packet_forwarder'
rm -f obj/*.o
rm -f lora_pkt_fwd
make[1]: Leaving directory '/home/radxa/sx1302_hal/packet_forwarder'
make clean -e -C util_net_downlink
make[1]: Entering directory '/home/radxa/sx1302_hal/util_net_downlink'
Makefile:27: /\/\/\/  Building in 'release' mode \/\/\/\
rm -f obj/*.o
rm -f net_downlink
make[1]: Leaving directory '/home/radxa/sx1302_hal/util_net_downlink'
make clean -e -C util_chip_id
make[1]: Entering directory '/home/radxa/sx1302_hal/util_chip_id'
Makefile:27: /\/\/\/  Building in 'release' mode \/\/\/\
rm -f obj/*.o
rm -f chip_id
make[1]: Leaving directory '/home/radxa/sx1302_hal/util_chip_id'
make clean -e -C util_boot
make[1]: Entering directory '/home/radxa/sx1302_hal/util_boot'
Makefile:27: /\/\/\/  Building in 'release' mode \/\/\/\
rm -f obj/*.o
rm -f boot
make[1]: Leaving directory '/home/radxa/sx1302_hal/util_boot'
make clean -e -C util_spectral_scan
make[1]: Entering directory '/home/radxa/sx1302_hal/util_spectral_scan'
Makefile:27: /\/\/\/  Building in 'release' mode \/\/\/\
rm -f obj/*.o
rm -f spectral_scan
make[1]: Leaving directory '/home/radxa/sx1302_hal/util_spectral_scan'
make all -e -C libtools
make[1]: Entering directory '/home/radxa/sx1302_hal/libtools'
mkdir -p obj
gcc -c -O2 -Wall -Wextra -std=c99 -Iinc -I. src/tinymt32.c -o obj/tinymt32.o
In file included from inc/tinymt32.h:19,
                 from src/tinymt32.c:16:
/usr/lib/gcc/aarch64-linux-gnu/10/include/stdint.h:9:16: fatal error: stdint.h: No such file or directory
    9 | # include_next <stdint.h>
      |                ^~~~~~~~~~
compilation terminated.
make[1]: *** [Makefile:33: obj/tinymt32.o] Error 1`

Please suggest me what i can do.

Regards 
Hitesh

hiteshtechshslok, 2025-01-28:
Update, i was able to install it just needed to fix the dependency .
But how can i chaneg GPIO's used

mcoracin, 2025-02-04:
Hello,
The GPIOs used can be updated in the reset_lgw.sh script which is run when the packet forwarder starts:
https://github.com/Lora-net/sx1302_hal/blob/master/tools/reset_lgw.sh
This is valid only if you use the SPI version of the corecell board.
Best regards,
Michael

hiteshtechshslok, 2025-02-04:
Hello @mcoracin 

Thank you for the response, But does it impact on SPI pins. like for orange pi pins may be different for SPI. So does Linux system takes default SPI pins in this as well. 

Regards, 
Hitesh

mcoracin, 2025-02-04:
No, it doesn't impact SPI pins.
The point is that if your sx1302 board (corecell) is the SPI version, you need to check if the GPIOs configured in the reset_lgw.sh script are correct for your hardware.
If you are using the USB version of the sx1302 corecell, the GPIO handling is done by the internal MCU so you don't need to worry about.

hiteshtechshslok, 2025-02-04:
Thanks @mcoracin , i will try to connect now.

IoTopenTech, 2025-04-15:
Hi @hiteshtechshslok I've succeeded installing sx1302_hal on a Radxa Rock S0.
I have discovered that the spi must be changed to spidev2.0 in the util_chip_id/srd/chip_id.c file (and then make clean all), and also in the conf.json.sx1250.EU868 file.
Kind regards from Madrid.