Lora-net/sx1302_hal #138: Receiving only 1/3 of LoRa transmission

NCPlyn, 2025-02-09:
I have a RAK7391 board with RPi CM4 and a RAK2287 (SPI SX1302) as a basestation and then Heltec ESP32 LoRa v3 (SX1262) as a node.
On the RPi I'm running the loragw_hal_rx with this command: `./test_loragw_hal_rx -r 1250 -a 868.4 -b 868.5 -k 0 -m 0` and I have edited it only so the datarate is SF10 and bandwidth 125KHZ.
The SX1262 has same sf,bw, frequency at 868.5, coding rate 6, public sync word and power at 20.. nearly the max (overpowers FM station on RTLSDR) using RadioLib.

The problem I'm having is that only like 1/3 of the transmissions/packets get recieved (with rssi 150+), if I change on both the sf to 6 and bw to 250, then I get none... Where there can be a problem?

mcoracin, 2025-02-10:
When you say: "I have edited it only so the datarate is SF10 and bandwidth 125KHZ", what did you do exactly ?
The sx1302 has 8 channels with multi-sf, meaning that it can receive all SFs on all channels, but only with 125khz bandwidth. Then there is an extra channel which is capable of 1 SF only, but bandwidth can be configured.

So by default, when using test_loragw_hal_rx, you don't need to change anything to receive SF10/BW125.

When using the option "-m 0", I would recommend launching the app with:
`./test_loragw_hal_rx -r 1250 -a 867.5 -b 868.5 -k 0 -m 0`
This will enable channels at:
867.1 - all SF BW125
867.3 - all SF BW125
867.5 - all SF BW125
867.7 - all SF BW125
867.9 - all SF BW125
868.1 - all SF BW125
868.3 - all SF BW125
868.5 - all SF BW125
868.3 - SF7 BW250

On the sx1262 side, could you try with coding rate 1 (4/5) ?
Try different frequency based on the list above to see if it behaves always the same.

NCPlyn, 2025-02-10:
Oh I see, well then I have recloned the repo and only edited the `test_loragw_hal_rx.c` to printf the payload and `loragw_hal.c` to not interact with the temperature sensor as the RAK2287 doesn't have I2C. (Lines 1095,1224,1290)
Here is the code for the node: [https://hastebin.com/share/kosexadaqe.cpp](https://hastebin.com/share/kosexadaqe.cpp) (867.5,125,4/5,sf10)
I have used this command line arguments: `./test_loragw_hal_rx -r 1250 -a 867.5 -b 868.5 -k 0 -m 0 -o -215.4`

And after trying some of the frequencies, with either the antennas 1m from each other or right next to each other, I recieve one, then I don't, then I do again... like 40% reception rate.

Most of the times like this:
```
----- LoRa packet -----
  count_us: 9082112
  size:     17
  chan:     5
  status:   0x10
  datr:     10
  codr:     1
  rf_chain  0
  freq_hz   867500000
  snr_avg:  9.5
  rssi_chan:-42.4
  rssi_sig :-43.4
  crc:      0xE2EC
74 65 73 74 4D 53 47 3A 70 6B 74 6E 75 6D 3A 38 30
testMSG:pktnum:80
```
But sometimes "corrupted" like this on wrong frequency:
```
----- LoRa packet -----
  count_us: 6472076
  size:     17
  chan:     7
  status:   0x11
  datr:     10
  codr:     1
  rf_chain  0
  freq_hz   867900000
  snr_avg:  -12.0
  rssi_chan:-103.4
  rssi_sig :-111.4
  crc:      0x3BE4
74 65 73 74 4D 53 47 3A F6 6B 74 6E 45 AD 32 37 38
testMSG:▒ktnE▒278
```
Cheap RTLSDR screenshot: (the 0.02 shift can but doesnt have to be due to the rtlsdr)
[image]

mcoracin, 2025-02-11:
Hello,

I've never used the RAK2287 module for the gateway, but I think that your gateway configuration is fine.
I have more doubt on the RadioLib configuration. I've never used it, but in order to match with the gateway configuration, you should ensure the following:
- the LoRa packet CRC should be enabled
- the LoRa preamble size should be set to 8
- explicit LoRa header should be used

If your end-device is close to the gateway, you definitely don't need to set the output power to 20dBm. 0dBm is more than enough. If it doesn't receive it means that there is a configuration issue somewhere, but not linked to the power.

If you receive packets on the wrong frequency, it could be due to the high ouput power creating RF images that can be detected as a packet but with CRC error.

With RadioLib, there are some examples for LoRaWAN, maybe you can test those examples, because the configuration should be aligned with the gateway. Just ensure that the syncword used is 0x34 (public).

Best regards,
Michael

NCPlyn, 2025-03-03:
Sorry for long time no respose, life go in the the way.
My other questions:
- Does antenna matter? I'm using small PCB antenna taht was given with the concentrator on the gateway and stubby antenna for the transmitter.
- Does vertical/horizontal position of recievers and transmitter matter?
- What good price/value ratio antenna should I get for the gateway?

About the test with TTN, I have set up the heltec SX1262 with TTN code and the results are these: each line is TX with which gateway recieved it at which frequency:
`./test_loragw_hal_rx -r 1250 -a 867.5 -b 868.5 -k 0 -m 0 -o -215.4`
```
46:56 ttn     (867.9 sf8)
47:17     rak (867.5 sf8)
47:25     rak (868.1 sf8)
47:33 ttn     (867.7 sf9)
47:56 ttn rak (867.7 sf9)
48:17 ttn     (868.1 sf9)
48:37 ttn     (867.9 sf9)
48:57 ttn     (868.5 sf8)
49:17 ttn     (868.5 sf8)
49:38         (867.5)
49:46 ttn rak (868.5 sf8)
49:59 ttn     (868.5 sf8)
50:19 ttn rak (867.3 sf8)
50:40 ttn rak (867.3 sf7)
51:00 ttn rak (867.9 sf7)
51:08     rak (867.5 sf7)
51:16         (867.3)
51:25     rak (868.1 sf8)
51:40 ttn rak (867.3 sf8)
51:49 ttn     (868.3)
52:00 ttn     (865.5 sf8)
52:20     rak (867.5 sf8)
52:29 ttn     (868.1 sf8)
```

Here are the log files I was coming from:

[ttnoutput.json](https://github.com/user-attachments/files/19061271/ttnoutput.json)
[rakoutput.txt](https://github.com/user-attachments/files/19061273/rakoutput.txt)
[heltecoutput.txt](https://github.com/user-attachments/files/19061272/heltecoutput.txt)

Another few questions:
- Should I blindly sent 3 packets with the transmitter and hope atleast one got recieved?
- Can I mix the hal_rx & hal_tx codes together and have the gateway reply with some ACK packet?

Thanks in advace and sorry for any spelling mistakes, it's nearly midnight.